allangood/rtlamr2mqtt #241: Device found but returning blank device info

klarrieu, 2024-08-26:
This was running fine for me but it recently stopped working. It looks like it's finding the device, 

```
rtlamr2mqtt  | [2024-08-26 19:24:56] RTL SDR Device 0bda:2838 found on USB port 001:003 - Index: 0
```

but then returning blank data:

```
rtlamr2mqtt  | [2024-08-26 19:25:01] RTL_TCP: Found 1 device(s):
rtlamr2mqtt  |   0:  , , SN: 
```

which then appears to lead to the following (not sure if that's the reason):

```
rtlamr2mqtt  | Using device 0: Generic RTL2832U OEM
rtlamr2mqtt  | usb_open error -4
rtlamr2mqtt  | Failed to open rtlsdr device #0.
```

The error code appears to correspond to `LIBUSB_ERROR_NO_DEVICE`.

`rtlamr` works fine for me if I run it outside the docker container, so I'm not sure what is going wrong here.

Here is my `rtlamr2mqtt.yaml`:
```
general:
  sleep_for: 5
  verbosity: debug

mqtt:
  host: localhost
  user:
  password:
  ha_autodiscovery: true
  ha_autodiscovery_topic: homeassistant

meters:
  - id: <redacted>
    protocol: scm+
    name: meter_water
    format: "#####"
    unit_of_measurement: "ft³"
    device_class: water
    state_class: total_increasing
```

Any advice is much appreciated!

allangood, 2025-05-30:
This looks like a hardware problem.
Sometime this happens on my machine - I have a cheap RTL device. The only way to fix it is to power my machine off, and power it again.

The USB reset can't fix this, neither removing the USB device and plug it in again.

allangood, 2025-05-30:
I will close this as preparation for the next release.

closed 2025-05-30 by allangood