allangood/rtlamr2mqtt #244: Seeing devices in log but not in MQTT

binaryhellstorm, 2024-10-04:
I am using this in Home Assistant on a Raspberry Pi 4, with the rtlamr2mqtt installed from the add-on store. 

I see what looks like the SDR picking up meters:
```
[2024-10-04 16:01:02] Using "/data/options.json" config file
[2024-10-04 16:01:02] MQTT Host defined in config file. Ignoring Supervisor Configuration...
[2024-10-04 16:01:02] Starting in LISTEN ONLY Mode...
[2024-10-04 16:01:02] Using "/data/options.json" config file
[2024-10-04 16:01:02] MQTT Host defined in config file. Ignoring Supervisor Configuration...
[2024-10-04 16:01:02] Configured MQTT sender:
[2024-10-04 16:01:02]  > hostname => 192.168.X.XXX
[2024-10-04 16:01:02]  > port => 1883
[2024-10-04 16:01:02]  > username => XXX
[2024-10-04 16:01:02]  > client_id => rtlamr2mqtt
[2024-10-04 16:01:02]  > base_topic => rtlamr
[2024-10-04 16:01:02]  > availability_topic => rtlamr/status
[2024-10-04 16:01:02]  > tls => None
[2024-10-04 16:01:02] Starting rtl_tcp with /usr/bin/rtl_tcp 
[2024-10-04 16:01:04] Starting rtlamr with ['/usr/bin/rtlamr', '-msgtype=all', '-format=json']
[2024-10-04 16:01:04] You should see all utility meters after this line:
[2024-10-04 16:01:04] 16:01:04.211173 decode.go:45: CenterFreq: 912600155
[2024-10-04 16:01:04] 16:01:04.211672 decode.go:46: SampleRate: 2359296
[2024-10-04 16:01:04] 16:01:04.211702 decode.go:47: DataRate: 32768
[2024-10-04 16:01:04] 16:01:04.211719 decode.go:48: ChipLength: 72
[2024-10-04 16:01:04] 16:01:04.211732 decode.go:49: PreambleSymbols: 32
[2024-10-04 16:01:04] 16:01:04.211744 decode.go:50: PreambleLength: 4608
[2024-10-04 16:01:04] 16:01:04.211756 decode.go:51: PacketSymbols: 736
[2024-10-04 16:01:04] 16:01:04.211814 decode.go:52: PacketLength: 105984
[2024-10-04 16:01:04] 16:01:04.211924 decode.go:59: Protocols: scm+,idm,r900,scm
[2024-10-04 16:01:04] 16:01:04.212010 decode.go:60: Preambles: 00000000000000001110010101100100,111110010101001100000,0001011010100011,01010101010101010001011010100011
[2024-10-04 16:01:04] 16:01:04.212084 main.go:124: GainCount: 29
```
I also see the username that's configured in the config hitting the Mosquito broker on HA

```
2024-10-04 15:50:21: New connection from XXX.30.33.5:56721 on port 1883.
2024-10-04 15:50:21: New client connected from XXX.30.33.5:56721 as rtlamr2mqtt (p2, c1, k60, u'addons').
2024-10-04 15:50:22: Client rtlamr2mqtt disconnected.
```

But if I go into MQTT devices, I see nothing, and when I listen to the topic rtlamr/debug I get nothing. I'm not sure what I'm doing wrong, I'm just trying to listen for available meters to get the info to configure mine.

hreed1985, 2024-11-26:
I am getting the same results. Did you ever find a solution?

0fxyzSa0vj, 2025-01-12:
> I am getting the same results. Did you ever find a solution?

Did you ever figure anything out? Getting exactly the same results. Been tinkering with it for several hours but no joy. Would love to get this working

hreed1985, 2025-01-23:
This shows up as an entity and not a device. This was part of my confusion. If you add this integration look for the entity and not device.

allangood, 2025-05-30:
I will close this as preparation for the next release.