allangood/rtlamr2mqtt #253: How to ask for help

allangood, 2025-05-05:
### The problem:

- I am getting burned trying to help all the "it doesn't work, please help" issues.
- I receive lots of "issues" that are actually not issues, but request on how to run or configure.
- I need to make it clear what is an issue and how to report an issue in a way I can help.

### What I can do to fix this:
Create template for issues.
Get better documentation

PrplHaz4, 2025-05-07:
You can also label things for community support - there should be no expectation you will personally respond to every issue - esp when a good number of issues can be resolved by reading the docs or waiting for your meter to be found.  With a rewrite in the works there's likely to be an uptick in issues overall - so triage will be important...

allangood, 2025-05-07:
These are good points @PrplHaz4 . 

Github has the "Discussion" section and the issues. I would like to redirect all "support requests" to the Discussion session and keep only issues in this section.

Another point on this are the "can you add support to meter X, Y or Z?"
As I am not the maintainer of the `rtlamr` , but my code is just a wrap around it to collect the data and send to mqtt.

I have to make sure this is clearly stated in the `README.md` file.