allangood/rtlamr2mqtt #254: Help needed

allangood, 2025-05-05:
### Help needed
Hello all.

I am getting back to this project and I need some help.

I moved to a house that doesn't have the meter transmitters. I can't test if it working anymore.

Could you please add a comment in this issue with an output from your device?

### How to do it?

Run these commands:
```
# docker run -ti --device /dev/bus/usb:/dev/bus/usb --entrypoint /bin/bash allangood/rtlamr2mqtt
rtl_tcp &
rtlamr -format=json -msgtype=all -duration=60s
```

This will have an output similar to this:
```
root@bf32335ba111:/# rtl_tcp &
[1] 7
root@bf32335ba111:/# Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Tuned to 100000000 Hz.
listening...
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).

root@bf32335ba111:/# rtlamr -format=json -msgtype=all -duration=60s
client accepted! localhost 59296
Allocating 15 zero-copy buffers
set freq 912380000
20:49:41.035317 decode.go:45: CenterFreq: 912380000
20:49:41.037656 decode.go:46: SampleRate: 2359296
20:49:41.037675 decode.go:47: DataRate: 32768
20:49:41.037684 decode.go:48: ChipLength: 72
20:49:41.037694 decode.go:49: PreambleSymbols: 32
20:49:41.037705 decode.go:50: PreambleLength: 4608
20:49:41.037716 decode.go:51: PacketSymbols: 736
20:49:41.037728 decode.go:52: PacketLength: 105984
20:49:41.037742 decode.go:59: Protocols: scm,scm+,idm,r900
20:49:41.037758 decode.go:60: Preambles: 0001011010100011,01010101010101010001011010100011,00000000000000001110010101100100,111110010101001100000
20:49:41.037781 main.go:124: GainCount: 29
set sample rate 2359296
set gain mode 0

```

Thank you in advance!

reypm, 2025-05-05:
here is my output ...

```
root@testing-server:/opt/docker/rtlamr2mqtt# docker run -ti --device /dev/bus/usb:/dev/bus/usb --entrypoint /bin/bash allangood/rtlamr2mqtt

root@f3bec803cb31:/# rtl_tcp &
[1] 7
root@f3bec803cb31:/# Found 1 device(s):
  0:  Generic, RTL2832U, SN: 77771111153705700

Using device 0: Generic RTL2832U
rtl_tcp Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Tuned to 100000000 Hz.
listening...
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).

root@f3bec803cb31:/# rtlamr -format=json -msgtype=all -duration=60s
client accepted! localhost 40854
Allocating 15 zero-copy buffers
set freq 912380000
21:14:30.923794 decode.go:45: CenterFreq: 912380000
21:14:30.923958 decode.go:46: SampleRate: 2359296
21:14:30.923974 decode.go:47: DataRate: 32768
21:14:30.923980 decode.go:48: ChipLength: 72
21:14:30.923986 decode.go:49: PreambleSymbols: 32
21:14:30.923997 decode.go:50: PreambleLength: 4608
21:14:30.924006 decode.go:51: PacketSymbols: 736
21:14:30.924015 decode.go:52: PacketLength: 105984
21:14:30.924027 decode.go:59: Protocols: scm,scm+,idm,r900
21:14:30.924046 decode.go:60: Preambles: 111110010101001100000,0001011010100011,01010101010101010001011010100011,00000000000000001110010101100100
21:14:30.924065 main.go:124: GainCount: 29
set sample rate 2359296
set gain mode 0
```

allangood, 2025-05-05:
Thank you @reypm for you quick reply!
But this output doesn't have any meter reading.
Could you try another shot with out the `-duration=60s`, please?

reypm, 2025-05-05:
sure, here is the output:

```
root@2ba6bdae7f60:/# rtlamr -format=json -msgtype=all -duration=60s
client accepted! localhost 56348
Allocating 15 zero-copy buffers
set freq 912600155
21:25:03.790381 decode.go:45: CenterFreq: 912600155
21:25:03.790583 decode.go:46: SampleRate: 2359296
21:25:03.790600 decode.go:47: DataRate: 32768
21:25:03.790614 decode.go:48: ChipLength: 72
21:25:03.790621 decode.go:49: PreambleSymbols: 32
21:25:03.790634 decode.go:50: PreambleLength: 4608
21:25:03.790641 decode.go:51: PacketSymbols: 736
21:25:03.790659 decode.go:52: PacketLength: 105984
21:25:03.790675 decode.go:59: Protocols: idm,r900,scm,scm+
21:25:03.790684 decode.go:60: Preambles: 01010101010101010001011010100011,00000000000000001110010101100100,111110010101001100000,0001011010100011
21:25:03.790703 main.go:124: GainCount: 29
set sample rate 2359296
set gain mode 0
{"Time":"2025-05-05T21:25:04.891578823Z","Offset":0,"Length":0,"Type":"IDM","Message":{"Preamble":1431639715,"PacketTypeID":28,"PacketLength":92,"HammingCode":198,"ApplicationVersion":4,"ERTType":7,"ERTSerialNumber":59897589,"ConsumptionIntervalCount":76,"ModuleProgrammingState":188,"TamperCounters":"AwIAcw4A","AsynchronousCounters":0,"PowerOutageFlags":"AAAAAAAA","LastConsumptionCount":1978208,"DifferentialConsumptionIntervals":[26,26,27,26,26,24,24,24,24,24,26,26,26,27,26,26,26,26,24,23,48,23,24,25,25,24,25,24,25,25,23,23,22,23,23,23,24,25,25,25,25,25,26,23,23,22,23],"TransmitTimeOffset":3911,"SerialNumberCRC":43319,"PacketCRC":49515}}
{"Time":"2025-05-05T21:25:06.548266268Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":65320502,"Type":7,"TamperPhy":1,"TamperEnc":2,"Consumption":7621974,"ChecksumVal":48922}}
{"Time":"2025-05-05T21:25:08.660603188Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":59897592,"Type":7,"TamperPhy":3,"TamperEnc":1,"Consumption":206082,"ChecksumVal":20730}}
{"Time":"2025-05-05T21:25:10.905527969Z","Offset":0,"Length":0,"Type":"R900","Message":{"ID":1850274558,"Unkn1":163,"NoUse":0,"BackFlow":0,"Consumption":4555831,"Unkn3":0,"Leak":2,"LeakNow":0}}
{"Time":"2025-05-05T21:25:11.001486809Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":61412301,"Type":7,"TamperPhy":0,"TamperEnc":1,"Consumption":9480653,"ChecksumVal":8042}}
{"Time":"2025-05-05T21:25:11.959372062Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":59897589,"Type":7,"TamperPhy":3,"TamperEnc":2,"Consumption":1978226,"ChecksumVal":60151}}
{"Time":"2025-05-05T21:25:14.555435329Z","Offset":0,"Length":0,"Type":"IDM","Message":{"Preamble":1431639715,"PacketTypeID":28,"PacketLength":92,"HammingCode":198,"ApplicationVersion":4,"ERTType":7,"ERTSerialNumber":61419216,"ConsumptionIntervalCount":68,"ModuleProgrammingState":188,"TamperCounters":"BAUAdhEA","AsynchronousCounters":0,"PowerOutageFlags":"AAAAAAAA","LastConsumptionCount":9386150,"DifferentialConsumptionIntervals":[57,85,51,41,69,43,42,43,43,43,43,44,44,44,44,46,41,27,43,44,43,43,43,43,43,44,70,52,44,45,46,44,47,44,43,24,41,43,45,44,43,44,41,25,43,44,43],"TransmitTimeOffset":4068,"SerialNumberCRC":11642,"PacketCRC":20482}}
{"Time":"2025-05-05T21:25:15.387177726Z","Offset":0,"Length":0,"Type":"IDM","Message":{"Preamble":1431639715,"PacketTypeID":52,"PacketLength":92,"HammingCode":198,"ApplicationVersion":140,"ERTType":6,"ERTSerialNumber":61412301,"ConsumptionIntervalCount":164,"ModuleProgrammingState":184,"TamperCounters":"BAEAbQEA","AsynchronousCounters":0,"PowerOutageFlags":"AAAACAAA","LastConsumptionCount":1217440205,"DifferentialConsumptionIntervals":[7,6,70,70,6,6,22,2,358,5,6,5,24,2,262,136,13,87,6,6,6,6,6,5,6,5,129,390,6,6,7,6,7,8,5,6,7,6,6,7,7,6,6,22,7,6,22],"TransmitTimeOffset":3938,"SerialNumberCRC":40239,"PacketCRC":58148}}
{"Time":"2025-05-05T21:25:16.459564528Z","Offset":0,"Length":0,"Type":"IDM","Message":{"Preamble":1431639715,"PacketTypeID":28,"PacketLength":92,"HammingCode":198,"ApplicationVersion":4,"ERTType":7,"ERTSerialNumber":59897589,"ConsumptionIntervalCount":76,"ModuleProgrammingState":188,"TamperCounters":"AwIAcw4A","AsynchronousCounters":0,"PowerOutageFlags":"AAAAAAAA","LastConsumptionCount":1978208,"DifferentialConsumptionIntervals":[26,26,27,26,26,24,24,24,24,24,26,26,26,27,26,26,26,26,24,23,48,23,24,25,25,24,25,24,25,25,23,23,22,23,23,23,24,25,25,25,25,25,26,23,23,22,23],"TransmitTimeOffset":4096,"SerialNumberCRC":43319,"PacketCRC":38431}}
{"Time":"2025-05-05T21:25:18.0592798Z","Offset":0,"Length":0,"Type":"IDM","Message":{"Preamble":1431639715,"PacketTypeID":28,"PacketLength":92,"HammingCode":198,"ApplicationVersion":4,"ERTType":7,"ERTSerialNumber":59897592,"ConsumptionIntervalCount":83,"ModuleProgrammingState":188,"TamperCounters":"AwEAgQ0A","AsynchronousCounters":0,"PowerOutageFlags":"AAAAAAAA","LastConsumptionCount":206053,"DifferentialConsumptionIntervals":[30,17,37,38,17,32,41,19,29,45,30,19,42,35,17,37,41,19,27,43,36,17,31,40,16,27,41,28,17,37,37,17,25,42,22,18,41,32,17,31,39,17,24,42,20,18,40],"TransmitTimeOffset":4091,"SerialNumberCRC":30874,"PacketCRC":37394}}
{"Time":"2025-05-05T21:25:18.404352806Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":61412301,"Type":7,"TamperPhy":0,"TamperEnc":1,"Consumption":9480653,"ChecksumVal":8042}}
{"Time":"2025-05-05T21:25:19.118764497Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":59897592,"Type":7,"TamperPhy":3,"TamperEnc":1,"Consumption":206082,"ChecksumVal":20730}}
[... cut ...] 7,31,40,16,27,41,28,17,37,37,17,25,42,22,18,41,32,17,31,39,17,24,42,20,18,40],"TransmitTimeOffset":4766,"SerialNumberCRC":30874,"PacketCRC":49320}}
{"Time":"2025-05-05T21:26:03.675556939Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":59897587,"Type":7,"TamperPhy":2,"TamperEnc":1,"Consumption":2691233,"ChecksumVal":33014}}
21:26:03.791181 main.go:341: Time Limit Reached: 1m0.000469378s
comm recv bye
Signal caught, exiting!
root@2ba6bdae7f60:/# worker cond timeout
Signal caught, exiting!
```

PrplHaz4, 2025-05-06:
Thanks for keeping this project alive - been running without any issues for a few years on my end!

Here's another capture:

```
root@f915459c9e7f:/# rtl_tcp & rtlamr -format=json -msgtype=all -duration=60s
[2] 17
Found 1 device(s):
client accepted! localhost 35218
set freq 912380000
Allocating 15 zero-copy buffers
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
usb_claim_interface error -6
Failed to open rtlsdr device #0.
03:41:12.172775 decode.go:45: CenterFreq: 912380000
03:41:12.200006 decode.go:46: SampleRate: 2359296
03:41:12.200026 decode.go:47: DataRate: 32768
03:41:12.200032 decode.go:48: ChipLength: 72
03:41:12.200041 decode.go:49: PreambleSymbols: 32
03:41:12.200046 decode.go:50: PreambleLength: 4608
03:41:12.200054 decode.go:51: PacketSymbols: 736
03:41:12.200071 decode.go:52: PacketLength: 105984
03:41:12.200079 decode.go:59: Protocols: scm,scm+,idm,r900
03:41:12.200088 decode.go:60: Preambles: 0001011010100011,01010101010101010001011010100011,00000000000000001110010101100100,111110010101001100000
03:41:12.200098 main.go:124: GainCount: 29
set sample rate 2359296
set gain mode 0
{"Time":"2025-05-06T03:41:13.053848684Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:13.111963254Z","Offset":0,"Length":0,"Type":"R900","Message":{"ID":1572910054,"Unkn1":163,"NoUse":33,"BackFlow":1,"Consumption":1356645,"Unkn3":0,"Leak":1,"LeakNow":0}}
{"Time":"2025-05-06T03:41:13.457217286Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:13.605245894Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:16.466533437Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:16.624136077Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:16.777936396Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:17.223649323Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:18.242393469Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:18.495724913Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:18.637649037Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:19.662819092Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:20.514580934Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:21.67483134Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:21.966279772Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:23.287721799Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:23.831103322Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:25.126340833Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:25.277280045Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:25.523412595Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:25.673880114Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:26.132389346Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
{"Time":"2025-05-06T03:41:26.532853326Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398966,"ChecksumVal":46232}}
[... cut ...]
{"Time":"2025-05-06T03:42:12.076668252Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":47596270,"Type":5,"TamperPhy":0,"TamperEnc":0,"Consumption":2398968,"ChecksumVal":20552}}
03:42:12.204216 main.go:341: Time Limit Reached: 1m0.000566588s
comm recv bye
Signal caught, exiting!
[2]+  Exit 1                  rtl_tcp
root@f915459c9e7f:/# worker cond timeout
Signal caught, exiting!
all threads dead..
listening...
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
```

allangood, 2025-05-06:
Thank you very much @reypm and @PrplHaz4 , with these outputs I can build a mock script to keep going!

closed 2025-05-06 by allangood