allangood/rtlamr2mqtt #273: [BUG] Newest version will not boot in Docker

Jaxel, 2025-06-02:
**Describe the bug**
Upgraded to the newest version this morning. Now my docker instance keeps crashing.

```
general:
  sleep_for: 0
  verbosity: info
  tickle_rtl_tcp: false
  device_id: "0bda:2838"

mqtt:
  host: 172.17.0.1
  port: 1883
  user: homeassistant
  password: **********
  ha_autodiscovery: true
  ha_autodiscovery_topic: homeassistant
  base_topic: rtlamr2mqtt
  
custom_parameters:
  rtltcp: "-samplerate=2048000"
  rtlamr: "-unique=true -symbollength=32"

meters:
  - id: **********
    protocol: scm+
    name: florida_city_gas_**********
    icon: mdi:fire-circle
    format: "######.##"
    unit_of_measurement: "CCF"
    state_class: total_increasing
    device_class: gas
  - id: **********
    protocol: r900
    name: city_of_psl_**********
    icon: mdi:water-circle
    format: "######.#"
    unit_of_measurement: "gal"
    state_class: total_increasing
    device_class: water
```

**Screenshots or logs**
```
2025/06/02 07:59:06	stderr	[2025-06-02 07:59:06,403] INFO:Starting rtlamr2mqtt 2025.6.1
2025/06/02 07:59:06	stderr	[2025-06-02 07:59:06,442] INFO:Config loaded successfully
2025/06/02 07:59:06	stderr	[2025-06-02 07:59:06,443] INFO:Connecting to MQTT broker at 172.17.0.1:1883
2025/06/02 07:59:06	stderr	[2025-06-02 07:59:06,474] INFO:Subscribing to homeassistant/status
2025/06/02 07:59:06	stderr	[2025-06-02 07:59:06,476] INFO:Publishing to homeassistant/device/**********/config: {"device": {"identifiers": "meter_**********", "name": "florida_city_gas_**********", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": **********, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.1", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"**********_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr2mqtt/**********/attributes", "unique_id": "**********_reading", "protocol": "scm+", "icon": "mdi:fire-circle", "format": "######.##", "unit_of_measurement": "CCF", "state_class": "total_increasing", "device_class": "gas"}, "**********_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "date", "value_template": "{{ value_json.lastseen }}", "unique_id": "**********_lastseen"}}, "state_topic": "rtlamr2mqtt/**********/state", "availability_topic": "rtlamr2mqtt/status", "qos": 1}
2025/06/02 07:59:06	stderr	[2025-06-02 07:59:06,476] INFO:Publishing to homeassistant/device/**********/config: {"device": {"identifiers": "meter_**********", "name": "city_of_psl_**********", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": **********, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.1", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"**********_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr2mqtt/**********/attributes", "unique_id": "**********_reading", "protocol": "r900", "icon": "mdi:water-circle", "format": "######.#", "unit_of_measurement": "gal", "state_class": "total_increasing", "device_class": "water"}, "**********_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "date", "value_template": "{{ value_json.lastseen }}", "unique_id": "**********_lastseen"}}, "state_topic": "rtlamr2mqtt/**********/state", "availability_topic": "rtlamr2mqtt/status", "qos": 1}
2025/06/02 07:59:07	stderr	[2025-06-02 07:59:07,477] INFO:Publishing to rtlamr2mqtt/status: online
2025/06/02 07:59:07	stderr	[2025-06-02 07:59:07,690] DEBUG:Reseting USB device: 0bda:2838
2025/06/02 07:59:07	stderr	Traceback (most recent call last):
2025/06/02 07:59:07	stderr	  File "/opt/venv/app/rtlamr2mqtt.py", line 394, in <module>
2025/06/02 07:59:07	stderr	    main()
2025/06/02 07:59:07	stderr	    ~~~~^^
2025/06/02 07:59:07	stderr	  File "/opt/venv/app/rtlamr2mqtt.py", line 300, in main
2025/06/02 07:59:07	stderr	    rtltcp = start_rtltcp(config)
2025/06/02 07:59:07	stderr	  File "/opt/venv/app/rtlamr2mqtt.py", line 121, in start_rtltcp
2025/06/02 07:59:07	stderr	    usbutil.reset_usb_device(usb_id)
2025/06/02 07:59:07	stderr	    ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
2025/06/02 07:59:07	stderr	  File "/opt/venv/app/helpers/usb_utils.py", line 44, in reset_usb_device
2025/06/02 07:59:07	stderr	    busnum, devnum = [int(x) for x in usbdev.split(':')]
2025/06/02 07:59:07	stderr	                      ~~~^^^
2025/06/02 07:59:07	stderr	ValueError: invalid literal for int() with base 10: '0bda'
```

**Version Information:**
 - OS: Docker (Synology Container Manager)
 - Version 2025.6.1

**Additional context**
Downgraded to 2.3.4 and everything works.

allangood, 2025-06-02:
There is a change on how the `device_id` works.

https://github.com/allangood/rtlamr2mqtt?tab=readme-ov-file#multiple-rtl-devices

Jaxel, 2025-06-02:
That's not how my lsusb presents:
```
    |__1-2.4     0bda:2838:0100 00  2.00  480MBit/s 500mA 2IFs (RTLSDRBlog Blog V4 00000001)
```

allangood, 2025-06-02:
The problem seems to be the `-samplerate=2048000` parameter. It does not exists.
Change it to `-s 2048000`

Jaxel, 2025-06-02:
Alright. Updated and rebuilt. Started up with no errors or warnings.
```
2025/06/02 13:22:52	stderr	[2025-06-02 13:22:52,078] INFO:Starting rtlamr2mqtt 2025.6.1
2025/06/02 13:22:52	stderr	[2025-06-02 13:22:52,083] INFO:Config loaded successfully
2025/06/02 13:22:52	stderr	[2025-06-02 13:22:52,083] INFO:Connecting to MQTT broker at 172.17.0.1:1883
2025/06/02 13:22:52	stderr	[2025-06-02 13:22:52,111] INFO:Subscribing to homeassistant/status
2025/06/02 13:22:52	stderr	[2025-06-02 13:22:52,112] INFO:Publishing to homeassistant/device/**********/config: {"device": {"identifiers": "meter_**********", "name": "florida_city_gas_**********", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": **********, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.1", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"**********_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr2mqtt/**********/attributes", "unique_id": "**********_reading", "protocol": "scm+", "icon": "mdi:fire-circle", "format": "######.##", "unit_of_measurement": "CCF", "state_class": "total_increasing", "device_class": "gas"}, "**********_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "date", "value_template": "{{ value_json.lastseen }}", "unique_id": "**********_lastseen"}}, "state_topic": "rtlamr2mqtt/**********/state", "availability_topic": "rtlamr2mqtt/status", "qos": 1}
2025/06/02 13:22:52	stderr	[2025-06-02 13:22:52,112] INFO:Publishing to homeassistant/device/**********/config: {"device": {"identifiers": "meter_**********", "name": "city_of_psl_**********", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": **********, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.1", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"**********_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr2mqtt/**********/attributes", "unique_id": "**********_reading", "protocol": "r900", "icon": "mdi:water-circle", "format": "######.#", "unit_of_measurement": "gal", "state_class": "total_increasing", "device_class": "water"}, "**********_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "date", "value_template": "{{ value_json.lastseen }}", "unique_id": "**********_lastseen"}}, "state_topic": "rtlamr2mqtt/**********/state", "availability_topic": "rtlamr2mqtt/status", "qos": 1}
2025/06/02 13:22:53	stderr	[2025-06-02 13:22:53,113] INFO:Publishing to rtlamr2mqtt/status: online
2025/06/02 13:22:53	stderr	[2025-06-02 13:22:53,291] DEBUG:Reseting USB device: 001:006
2025/06/02 13:22:53	stderr	[2025-06-02 13:22:53,588] INFO:Starting RTL_TCP using "rtl_tcp -s 2048000 -d 0"
2025/06/02 13:22:53	stderr	Found 1 device(s):
2025/06/02 13:22:53	stderr	  0:  RTLSDRBlog, Blog V4, SN: 00000001
2025/06/02 13:22:53	stderr	
2025/06/02 13:22:53	stderr	Using device 0: Generic RTL2832U OEM
2025/06/02 13:22:53	stderr	Found Rafael Micro R828D tuner
2025/06/02 13:22:53	stderr	RTL-SDR Blog V4 Detected
2025/06/02 13:22:54	stderr	Tuned to 100000000 Hz.
```
We'll see if this all works. Sensors for "Last Seen" and "Reading" are still showing as "Unknown" though. My gas meter pulses once an hour. So maybe it will get updated when a pulse is detected.

trevnologies, 2025-06-02:
> I don't think the default frequency of 100000000 Hz is finding any meters in my neighborhood.
> 
> I was using the default frequency in 2.3.4. What was the default frequency there?

same and this is what i am wondering too. my logs look exactly like yours at this point, just a different USB Device ID is all

Jaxel, 2025-06-02:
After nearly an hour now, I'm getting nothing. It's connected to Mosquitto perfectly fine, and I can see it online in MQTT Explorer, but it never publishes any topics because it never receives any data.

allangood, 2025-06-07:
This is caused by #277 

I am closing this issue.

closed 2025-06-07 by allangood