allangood/rtlamr2mqtt #275: [BUG] No devices found/center frequency possibly not being set properly

apranda, 2025-06-02:
**Describe the bug**
No devices are being detected anymore and the frequency setting in the log suggests that the center frequency, which previously needed to be set to find my meters, isn't being used

I am using this Configuration:
```
general:
  sleep_for: 300
  verbosity: debug
  device_id: "002:003"
  rtltcp_host: 192.168.1.2:1234
  listen_only: false
  tickle_rtl_tcp: false
mqtt:
  host: 192.168.1.2
  port: 1883
  ha_autodiscovery_topic: homeassistant
  ha_status_topic: hass/status
  base_topic: rtlamr
  ha_autodiscovery: true
  tls_enabled: false
custom_parameters:
  rtltcp: "-s 2048000"
  rtlamr: "-unique=true -centerfreq=915000000"
meters:
  - id: 9661519
    protocol: scm
    name: gas_meter
    format: "######"
    unit_of_measurement: ft³
    icon: mdi:gas_cylinder
    device_class: gas
  - id: 4393811
    protocol: scm
    name: electric_meter
    format: "####.##"
    unit_of_measurement: kWh
    icon: mdi:electric_meter
    device_class: energy
    state_class: total

```

**Screenshots or logs**

[logs.txt](https://github.com/user-attachments/files/20552612/logs.txt)

**Version Information:**
 - OS: HAOS 2025.5.3
 - 2025.6.1

BigDogJonx, 2025-06-02:
I got it use the frequency by putting the following ingredients for the rtl_tcp line in the config, but still not getting my data unfortunately yet, so still playing with this.
`rtltcp: "-f 912600155 -s 2048000"`

BigDogJonx, 2025-06-02:
I can get it to start and use the frequency, but doesnt seem to be pulling any data.

```
[2025-06-02 11:52:11,963] INFO:Publishing to rtlamr/status: online
[2025-06-02 11:52:11,991] DEBUG:Reseting USB device: 003:002
[2025-06-02 11:52:12,275] INFO:Starting RTL_TCP using "rtl_tcp -f 912600155 -s 2048000 -d 0"
Found 1 device(s):
  0:  RTLSDRBlog, Blog V4, SN: 00000001
Using device 0: Generic RTL2832U OEM
Found Rafael Micro R828D tuner
RTL-SDR Blog V4 Detected
Tuned to 912600155 Hz.
```

Even tried to add more specific things under rtlamr but not sure its actually honoring any of it.
```
rtltcp: "-f 912600155 -s 2048000"
rtlamr: "-msgtype=scm+ -format=json -filterid=123456789 -unique=true"

```

allangood, 2025-06-02:
> I can get it to start and use the frequency, but doesnt seem to be pulling any data.
>
> ```
> [2025-06-02 11:52:11,963] INFO:Publishing to rtlamr/status: online
> [2025-06-02 11:52:11,991] DEBUG:Reseting [...]

This is covered here:
https://github.com/allangood/rtlamr2mqtt/discussions/270

BigDogJonx, 2025-06-02:
> 
> This is covered here: [#270](https://github.com/allangood/rtlamr2mqtt/discussions/270)

YAML configuration file doesnt have any options on how to get it to look at certain frequencies or lock into the meter.  You can see from above, I got it to look at the right center frequency, but it's still not pulling any data for some reason.  Before youd get much more where the rtlamr would have CenterFreq, SampleRate, DataRate, ChipLength, etc.  And then would also show the data of what meter is being filtered out.  

This is what the pervious version would show 

```
[2025-06-01 06:52:42] Resetting USB device: /dev/bus/usb/003/002
[2025-06-01 06:52:42] Reset sucessful.
[2025-06-01 06:52:42] Sending message to MQTT:
[2025-06-01 06:52:42]  > topic => rtlamr/status
[2025-06-01 06:52:42]  > payload => online
[2025-06-01 06:52:42]  > retain => True
[2025-06-01 06:52:42] Trying to start RTL_TCP: /usr/bin/rtl_tcp  -s 2048000
[2025-06-01 06:52:42] RTL_TCP started with PID 179702
[2025-06-01 06:52:47] RTL_TCP is ready to receive connections!
[2025-06-01 06:52:47] Trying to start RTLAMR: /usr/bin/rtlamr -msgtype=scm+ -format=json -filterid=12345678 -unique=true
[2025-06-01 06:52:47] RTLAMR started with PID 179707
[2025-06-01 06:52:47] 06:52:47.585717 decode.go:45: CenterFreq: 912600155
[2025-06-01 06:52:47] 06:52:47.585860 decode.go:46: SampleRate: 2359296
[2025-06-01 06:52:47] 06:52:47.585867 decode.go:47: DataRate: 32768
[2025-06-01 06:52:47] 06:52:47.585872 decode.go:48: ChipLength: 72
[2025-06-01 06:52:47] 06:52:47.585876 decode.go:49: PreambleSymbols: 16
[2025-06-01 06:52:47] 06:52:47.585882 decode.go:50: PreambleLength: 2304
[2025-06-01 06:52:47] 06:52:47.585886 decode.go:51: PacketSymbols: 128
[2025-06-01 06:52:47] 06:52:47.585895 decode.go:52: PacketLength: 18432
[2025-06-01 06:52:47] 06:52:47.585901 decode.go:59: Protocols: scm+
[2025-06-01 06:52:47] 06:52:47.585909 decode.go:60: Preambles: 0001011010100011
[2025-06-01 06:52:47] 06:52:47.585914 main.go:124: GainCount: 29
[2025-06-01 06:53:37] {"Time":"2025-06-01T06:53:37.561669561-04:00","Offset":0,"Length":0,"Type":"SCM+","Message":{"FrameSync":5795,"ProtocolID":30,"EndpointType":156,"EndpointID": 12345678,"Consumption":3437430,"Tamper":2568,"PacketCRC":25723}}
[2025-06-01 06:53:37] Meter "12345678" - Consumption 3437430. Sending value to MQTT.
[2025-06-01 06:53:37] Sending message to MQTT:
[2025-06-01 06:53:37]  > topic => rtlamr/12345678/attributes
[2025-06-01 06:53:37]  > payload => {"Message Type": "SCM+", "FrameSync": 5795, "ProtocolID": 30, "EndpointType": 156, "EndpointID": 12345678, "Consumption": 3437430, "Tamper": 2568, "PacketCRC": 25723}
[2025-06-01 06:53:37]  > retain => True
[2025-06-01 06:53:37] Sending message to MQTT:
[2025-06-01 06:53:37]  > topic => rtlamr/12345678/state
[2025-06-01 06:53:37]  > payload => 3437430
[2025-06-01 06:53:37]  > retain => True
[2025-06-01 06:53:37] Sleep_for defined, time to sleep!
[2025-06-01 06:53:37] Terminating all subprocess...
[2025-06-01 06:53:37] Kill process called.
[2025-06-01 06:53:37] Killing RTL_TCP...
[2025-06-01 06:53:42] Killed.
[2025-06-01 06:53:42] Killing RTLAMR...
[2025-06-01 06:53:42] Killed in the first attempt.
[2025-06-01 06:53:42] Sleeping for 300 seconds, see you later...
```

Seems like the rtlamr isnt decoding with all this info, even after attempting to inject it in the configuration YAML.

allangood, 2025-06-02:
Please, stop trying to hijack this issue without filling the proper information.
I can't guess the logs nor the configuration from your computer.

Thanks.

apranda, 2025-06-02:
> Also, could you confirm that your RTL_TCP server is running in this address?
> 
> ```
>   rtltcp_host: 192.168.1.2:1234
> ```
> 
> My guess is that you are using RTL_TCP as MQTT address. If you don't have a remote RTL_TCP Server, remove this line

I can't save the configuration with this line removed.

nar1117, 2025-06-03:
Same issue as OP here - add-on starts, connects to broker, publishes message, then launches rtl_tcp with apparently the correct flags, but logs show the same "Tuned to 100000000 Hz" - then nothing.

Nhscan, 2025-06-03:
Just for testing and a sanity check, I installed a fork that hasn't been updated to the latest rewrite — and bam, everything is back up and working great. I'll stick with this fork until the bugs are worked out. This is the fork I used: https://github.com/CertifiedDIY/rtlamr2mqtt_RTL-SDR-Blog

allangood, 2025-06-03:
> Just for testing and a sanity check, I installed a fork that hasn't been updated to the latest rewrite — and bam, everything is back up and working great. I'll stick with this fork until the bugs are worked out. This is the fork I used: https://github.com/CertifiedDIY/rtlamr2mqtt_RTL-SDR-Blog

I have no control or any relationship with this reposition. Use at your own risk.

dropte, 2025-06-04:
I'll add my +1 to this.  Here's my config and logs. Meter IDs have been redacted. 

```
general:
  sleep_for: 300
  verbosity: debug
  device_id: "0"
  rtltcp_host: 127.0.0.1:1234
  listen_only: false
  tickle_rtl_tcp: false
mqtt:
  host: core-mosquitto
  port: 1883
  ha_autodiscovery_topic: homeassistant
  ha_status_topic: hass/status
  base_topic: rtlamr
  ha_autodiscovery: true
  tls_enabled: false
custom_parameters:
  rtltcp: "-s 2048000"
  rtlamr: "-unique=true"
meters:
  - id: 12345678
    protocol: idm
    name: house_electric_meter
    format: "#####.##"
    unit_of_measurement: kWh
    icon: mdi:gauge
    device_class: energy
  - id: 87654321
    protocol: scm
    name: house_gas_meter
    format: "#####.##"
    unit_of_measurement: ft³
    icon: mdi:gauge
    device_class: gas
```

And logs look like this:
```
[2025-06-04 12:54:01,505] CRITICAL:Signal 15 received.
[2025-06-04 12:54:02,863] INFO:Starting rtlamr2mqtt 2025.6.1
[2025-06-04 12:54:02,864] INFO:Config loaded successfully
[2025-06-04 12:54:02,864] INFO:Connecting to MQTT broker at core-mosquitto:1883
[2025-06-04 12:54:02,869] INFO:Subscribing to hass/status
[2025-06-04 12:54:02,870] INFO:Publishing to homeassistant/device/12345678/config: {"device": {"identifiers": "meter_12345678", "name": "house_electric_meter", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": 12345678, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.1", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"12345678_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr/12345678/attributes", "unique_id": "12345678_reading", "protocol": "idm", "format": "#####.##", "unit_of_measurement": "kWh", "icon": "mdi:gauge", "device_class": "energy"}, "12345678_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "date", "value_template": "{{ value_json.lastseen }}", "unique_id": "12345678_lastseen"}}, "state_topic": "rtlamr/12345678/state", "availability_topic": "rtlamr/status", "qos": 1}
[2025-06-04 12:54:02,870] INFO:Publishing to homeassistant/device/87654321/config: {"device": {"identifiers": "meter_87654321", "name": "house_gas_meter", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": 87654321, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.1", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"87654321_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr/87654321/attributes", "unique_id": "87654321_reading", "protocol": "scm", "format": "#####.##", "unit_of_measurement": "ft\u00b3", "icon": "mdi:gauge", "device_class": "gas"}, "87654321_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "date", "value_template": "{{ value_json.lastseen }}", "unique_id": "87654321_lastseen"}}, "state_topic": "rtlamr/87654321/state", "availability_topic": "rtlamr/status", "qos": 1}
[2025-06-04 12:54:03,870] INFO:Publishing to rtlamr/status: online
[2025-06-04 12:54:03,900] DEBUG:Reseting USB device: 001:003
[2025-06-04 12:54:04,182] INFO:Starting RTL_TCP using "rtl_tcp -s 2048000 -d 0"
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Tuned to 100000000 Hz.
```

allangood, 2025-06-07:
This is caused by #277 

I am closing this issue.

closed 2025-06-07 by allangood