allangood/rtlamr2mqtt #283: after Rtlamr2mqtt update to 2025.6.1, no longer starts

dmd, 2025-06-04:
rtlamr2mqtt updated last night, and now doesn't start on boot. If I hit start manually, I get:

```
does not match regular expression (^0|^(0[xX])?[A-Fa-f0-9]+:(0[xX])?[A-Fa-f0-9]+$).
```

My config looks like this:

general:
```
sleep_for: 300
verbosity: debug
device_id: single
rtltcp_host: 127.0.0.1:1234
listen_only: false
tickle_rtl_tcp: false
```

mqtt:
```
host: 127.0.0.1
port: 1883
ha_autodiscovery_topic: homeassistant
ha_status_topic: hass/status
base_topic: rtlamr
ha_autodiscovery: true
tls_enabled: false
```

custom_parameters:
```
rtltcp: "-s 2048000"
rtlamr: "-unique=true"
```

meters:
```
- id: 60777417
  protocol: scm
  name: electric_meter
  format: "#####"
  unit_of_measurement: kWh
  icon: mdi:lightning-bolt
  device_class: energy
```

Any ideas?

dropte, 2025-06-04:
You need to find the usb device id. Output from lsusb or I found mine in the rpi system hardware info (mine is 001:003)

dropte, 2025-06-04:
It no longer allows mqtt to 127.0.0.1. Need to change it to the ip of your host, or the hostname of the mqtt broker (mine was `core-mosquitto` That’s covered in issue 270.

dmd, 2025-06-04:
By the way - presumably because I only have one RTL device - using just `device: 0` worked fine.