allangood/rtlamr2mqtt #288: [BUG] No MQTT messages from RTLAMR

allangood, 2025-06-07:
The rtlamr output is not a byte anymore.

Change this:
```
.stdout.read1().decode('utf-8')
```

To this:
```
stdout.readline().strip()
```

davidcoulson, 2025-06-07:
Still not working for me:
```

[2025-06-07 12:06:26,043] INFO:Starting rtlamr2mqtt 2025.6.3
[2025-06-07 12:06:26,044] DEBUG:Starting new HTTP connection (1): supervisor:80
[2025-06-07 12:06:26,046] DEBUG:http://supervisor:80 "GET /services/mqtt HTTP/1.1" 200 214
[2025-06-07 12:06:26,046] INFO:Config loaded successfully
[2025-06-07 12:06:26,046] INFO:Connecting to MQTT broker at core-mosquitto:1883
[2025-06-07 12:06:26,046] INFO:Subscribing to homeassistant/status
[2025-06-07 12:06:26,047] INFO:Publishing to homeassistant/device/102845955/config: {"device": {"identifiers": "meter_102845955", "name": "gas_meter", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": 102845955, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.3", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"102845955_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr/102845955/attributes", "unique_id": "102845955_reading", "protocol": "scm+", "format": "#####.##", "unit_of_measurement": "CCF", "icon": "mdi:meter-gas", "device_class": "gas"}, "102845955_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "date", "value_template": "{{ value_json.lastseen }}", "unique_id": "102845955_lastseen"}}, "state_topic": "rtlamr/102845955/state", "availability_topic": "rtlamr/status", "qos": 1}
[2025-06-07 12:06:27,047] INFO:Publishing to rtlamr/status: online
[2025-06-07 12:06:27,047] INFO:Using remote RTL_TCP server at 10.2.3.11:1234
[2025-06-07 12:06:27,248] INFO:Starting RTLAMR using: /usr/bin/rtlamr -format=json -server=10.2.3.11:1234 -unique=true -filterid=102845955 -msgtype=scm+
[2025-06-07 12:06:32,064] DEBUG:12:06:32.064504 decode.go:45: CenterFreq: 912600155
[2025-06-07 12:06:32,064] DEBUG:12:06:32.064632 decode.go:46: SampleRate: 2359296
[2025-06-07 12:06:32,064] DEBUG:12:06:32.064636 decode.go:47: DataRate: 32768
[2025-06-07 12:06:32,065] DEBUG:12:06:32.064639 decode.go:48: ChipLength: 72
[2025-06-07 12:06:32,065] DEBUG:12:06:32.064643 decode.go:49: PreambleSymbols: 16
[2025-06-07 12:06:32,065] DEBUG:12:06:32.064646 decode.go:50: PreambleLength: 2304
[2025-06-07 12:06:32,065] DEBUG:12:06:32.064651 decode.go:51: PacketSymbols: 128
[2025-06-07 12:06:32,065] DEBUG:12:06:32.064660 decode.go:52: PacketLength: 18432
[2025-06-07 12:06:32,065] DEBUG:12:06:32.064663 decode.go:59: Protocols: scm+
[2025-06-07 12:06:32,065] DEBUG:12:06:32.064672 decode.go:60: Preambles: 0001011010100011
[2025-06-07 12:06:32,065] DEBUG:12:06:32.064675 main.go:125: GainCount: 29
[2025-06-07 12:06:36,273] DEBUG:{"Time":"2025-06-07T12:06:36.272785767-04:00","Offset":0,"Length":49152,"Type":"SCM+","Message":{"FrameSync":5795,"ProtocolID":30,"EndpointType":188,"EndpointID":102845955,"Consumption":313422,"Tamper":3848,"PacketCRC":32316}}
```

DaftHonk, 2025-06-07:
#292 also covers this.

allangood, 2025-06-07:
Can someone help me to troubleshoot this?
On my end, I cannot reproduce the problem, mostly because I don't have a way to test with real devices anymore.

I would like to see the messages inside MQTT, like this:

[image]

Here, I can see all messages:
`homeassistant/device/<meter_id>/config`
and
`rtlamr/<meter_id>/state`
`rtlamr/<meter_id>/attribute`

DaftHonk, 2025-06-07:
As you can see, mine are not being created at all. I have no `rtlamr` topic at all and no sensors created (the ones visible are from zigbee2mqtt).
[image]

allangood, 2025-06-07:
Something wrong with your topics. 
Could you post your config please?

You are not supposed to have:
`homeassistant/sensor/rtlamr/power_meter`

amordicus, 2025-06-07:
> allancassaro
>  left a comment
> (allangood/rtlamr2mqtt#288)
> @allangood <https://github.com/allangood> I've put my information in the closed issue 293! Would you like me to copy it here? I went too fast.
>
> Yes, please 🙏

No problem, done. Are you helping allangood?  Let me know if I can help.

JohnyBeGood, 2025-06-07:
After the update, docker logs show that endpoint 314XX601 the latest `Consumption":7690` but MQTT topics not updating:

`[2025-06-07 15:56:05,475] INFO:Starting rtlamr2mqtt 2025.6.3
[2025-06-07 15:56:05,487] INFO:Config loaded successfully
[2025-06-07 15:56:05,487] INFO:Connecting to MQTT broker at 192.168.1.105:1883
[2025-06-07 15:56:05,490] INFO:Subscribing to homeassistant/status
[2025-06-07 15:56:05,492] INFO:Publishing to homeassistant/device/314XX600/config: {"device": {"identifiers": "meter_314XX600", "name": "inside_meter_water", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": 314XX600, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.3", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"314XX600_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr/314XX600/attributes", "unique_id": "314XX600_reading", "protocol": "scm", "format": "####.##", "unit_of_measurement": "ft\u00b3", "icon": "mdi:water", "device_class": "water", "state_class": "total_increasing"}, "314XX600_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "date", "value_template": "{{ value_json.lastseen }}", "unique_id": "314XX600_lastseen"}}, "state_topic": "rtlamr/314XX600/state", "availability_topic": "rtlamr/status", "qos": 1}
[2025-06-07 15:56:05,492] INFO:Publishing to homeassistant/device/314XX601/config: {"device": {"identifiers": "meter_314XX601", "name": "outside_meter_water", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": 314XX601, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.3", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"314XX601_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr/314XX601/attributes", "unique_id": "314XX601_reading", "protocol": "scm", "format": "####.##", "unit_of_measurement": "ft\u00b3", "icon": "mdi:water", "device_class": "water", "state_class": "total_increasing"}, "314XX601_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "date", "value_template": "{{ value_json.lastseen }}", "unique_id": "314XX601_lastseen"}}, "state_topic": "rtlamr/314XX601/state", "availability_topic": "rtlamr/status", "qos": 1}
[2025-06-07 15:56:06,496] INFO:Publishing to rtlamr/status: online
[2025-06-07 15:56:12,374] DEBUG:Reseting USB device: 003:004
[2025-06-07 15:56:12,375] INFO:Starting RTL_TCP using: /usr/bin/rtl_tcp -s 2048000 -d 0
[2025-06-07 15:56:12,423] DEBUG:Found 1 device(s):
[2025-06-07 15:56:12,444] DEBUG:0:  Nooelec, NESDR SMArt v5, SN: 00000001
[2025-06-07 15:56:12,450] DEBUG:Using device 0: Generic RTL2832U OEM
[2025-06-07 15:56:12,776] DEBUG:Found Rafael Micro R820T tuner
[2025-06-07 15:56:13,053] DEBUG:[R82XX] PLL not locked!
[2025-06-07 15:56:13,131] DEBUG:Tuned to 100000000 Hz.
[2025-06-07 15:56:13,160] DEBUG:listening...
[2025-06-07 15:56:13,160] INFO:RTL_TCP started!
[2025-06-07 15:56:13,361] INFO:Starting RTLAMR using: /usr/bin/rtlamr -format=json -server=127.0.0.1:1234 -unique=true -filterid=314XX600,314XX601 -msgtype=scm,scm
[2025-06-07 15:56:13,451] DEBUG:15:56:13.451090 decode.go:45: CenterFreq: 912600155
[2025-06-07 15:56:13,452] DEBUG:15:56:13.451253 decode.go:46: SampleRate: 2359296
[2025-06-07 15:56:13,452] DEBUG:15:56:13.451263 decode.go:47: DataRate: 32768
[2025-06-07 15:56:13,452] DEBUG:15:56:13.451268 decode.go:48: ChipLength: 72
[2025-06-07 15:56:13,452] DEBUG:15:56:13.451273 decode.go:49: PreambleSymbols: 21
[2025-06-07 15:56:13,452] DEBUG:15:56:13.451278 decode.go:50: PreambleLength: 3024
[2025-06-07 15:56:13,452] DEBUG:15:56:13.451283 decode.go:51: PacketSymbols: 96
[2025-06-07 15:56:13,452] DEBUG:15:56:13.451288 decode.go:52: PacketLength: 13824
[2025-06-07 15:56:13,453] DEBUG:15:56:13.451294 decode.go:59: Protocols: scm
[2025-06-07 15:56:13,453] DEBUG:15:56:13.451300 decode.go:60: Preambles: 111110010101001100000
[2025-06-07 15:56:13,453] DEBUG:15:56:13.451305 main.go:125: GainCount: 29
[2025-06-07 15:56:49,639] DEBUG:{"Time":"2025-06-07T15:56:49.639012345-07:00","Offset":0,"Length":40960,"Type":"SCM","Message":{"ID":314XX600,"Type":13,"TamperPhy":0,"TamperEnc":0,"Consumption":144691,"ChecksumVal":4864}}
[2025-06-07 15:57:45,862] DEBUG:{"Time":"2025-06-07T15:57:45.860588385-07:00","Offset":0,"Length":40960,"Type":"SCM","Message":{"ID":314XX601,"Type":13,"TamperPhy":0,"TamperEnc":0,"Consumption":7689,"ChecksumVal":6118}}
[2025-06-07 15:59:02,678] DEBUG:{"Time":"2025-06-07T15:59:02.677894985-07:00","Offset":0,"Length":40960,"Type":"SCM","Message":{"ID":314XX601,"Type":13,"TamperPhy":0,"TamperEnc":0,"Consumption":7690,"ChecksumVal":24483}}`

Jaxel, 2025-06-07:
Same issue for me. Meters are being read, but nothing is being sent to MQTT.

[image]

[image]

nrdufour, 2025-06-09:
Ah, I think I found the issue: the line 213 on the main py file, has the following:

                if 'set gain mode' in rtlamr_output:

The problem is "set gain mode" would never appear in rtlamr output, it's coming from rtl_tcp. Therefore, the code loops forever in the start_rtlamr function without doing anything else.

Changing it to:

                if 'GainCount' in rtlamr_output:

would fix the issue.

allangood, 2025-06-09:
> Ah, I think I found the issue: the line 213 on the main py file, has the following:
>
> ```
>             if 'set gain mode' in rtlamr_output:
> ```
>
> The problem is "set gain mode" would never appear in [...]

That was a really great catch!
Thank you for the feedback!

I am sending an update now.

allangood, 2025-06-09:
Change has been pushed to the `dev` branch.
@nrdufour , would you be able to test it, please?

allangood, 2025-06-09:
Thanks to @nrdufour  and [Linus law](https://en.wikipedia.org/wiki/Linus%27s_law)!

DayRunner131, 2025-06-10:
I appreciate the work you all did to track down the issue. However, HomeAssistant is not recognizing the new 2025.6.4 version update, possibly due to the ### before the version nomenclature. Were the hashes added purposely? I do not see any notes on it.

adamderuwe, 2025-06-11:
@DayRunner131  Looks like 2025.6.5 was just merged.  I was able to update the HA addon and am now getting MQTT messages published.

adamderuwe, 2025-06-11:
No entity shows up in Home Assistant now.

[image]

[image]

```
homeassistant/device/69177042/config

{
    "device": {
        "identifiers": "meter_69177042",
        "name": "electric_meter",
        "manufacturer": "RTLAMR2MQTT",
        "model": "Smart Meter",
        "sw_version": "1.0",
        "serial_number": 69177042,
        "hw_version": "1.0"
    },
    "origin": {
        "name": "2mqtt",
        "sw_version": "2025.6.5",
        "support_url": "https://github.com/allangood/rtlamr2mqtt"
    },
    "components": {
        "69177042_reading": {
            "platform": "sensor",
            "name": "Reading",
            "value_template": "{{ value_json.reading|float }}",
            "json_attributes_topic": "rtlamr/69177042/attributes",
            "unique_id": "69177042_reading",
            "protocol": "scm+",
            "format": "#####.##",
            "unit_of_measurement": "kWh",
            "icon": "mdi:gauge",
            "device_class": "energy"
        },
        "69177042_lastseen": {
            "platform": "sensor",
            "name": "Last Seen",
            "device_class": "timestamp",
            "value_template": "{{ value_json.lastseen }}",
            "unique_id": "69177042_lastseen"
        }
    },
    "state_topic": "rtlamr/69177042/state",
    "availability_topic": "rtlamr/status",
    "qos": 1
}
```

sdaltons1, 2025-06-11:
Thanks for the work on this!  

Just update to 6.5 and I'm back to the "does not match regular expression" error when I try to start the addon.

Config:

general:
  sleep_for: 300
  verbosity: info
  device_id: "0"
custom_parameters:
  rtltcp: "-s 2048000"
  rtlamr: "-unique=true"
mqtt:
  ha_autodiscovery_topic: homeassistant
  ha_status_topic: homeassistant/status
  base_topic: rtlamr
  host: 192.168.1.198
  port: 1883
  user: -----
  password: ------
  ha_autodiscovery: true
  tls_enabled: false
meters:
  - id: 110133043
    protocol: scm+
    name: gas_meter
    format: "###.##"
    unit_of_measurement: CCF
    icon: mdi:gas_canister
    device_class: gas
    state_class: total_increasing

closed 2025-06-11 by allangood