allangood/rtlamr2mqtt #293: 2025.6.3 hangs at Starting RTLAMR

keeperlink, 2025-06-07:
After updating to v. 2025.6.3 rtlamr2mqtt stops working. When I reversed it back to ver. 2.3.4 it works again.
Running on HA OS.

**LOG for ver. 2025.6.3:**
```
[2025-06-07 14:21:10,594] INFO:Starting rtlamr2mqtt 2025.6.3
[2025-06-07 14:21:10,596] DEBUG:Starting new HTTP connection (1): supervisor:80
[2025-06-07 14:21:10,599] DEBUG:http://supervisor:80 "GET /services/mqtt HTTP/1.1" 200 214
[2025-06-07 14:21:10,600] INFO:Config loaded successfully
[2025-06-07 14:21:10,601] INFO:Connecting to MQTT broker at core-mosquitto:1883
[2025-06-07 14:21:10,602] INFO:Subscribing to homeassistant/status
[2025-06-07 14:21:10,603] INFO:Publishing to homeassistant/device/1543497178/config: {"device": {"identifiers": "meter_1543497178", "name": "water_meter_home", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": 1543497178, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.3", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"1543497178_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr/1543497178/attributes", "unique_id": "1543497178_reading", "protocol": "r900", "format": "####.###", "unit_of_measurement": "m\u00b3", "icon": "mdi:water", "device_class": "water"}, "1543497178_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "date", "value_template": "{{ value_json.lastseen }}", "unique_id": "1543497178_lastseen"}}, "state_topic": "rtlamr/1543497178/state", "availability_topic": "rtlamr/status", "qos": 1}
[2025-06-07 14:21:10,603] INFO:Publishing to homeassistant/device/1541217672/config: {"device": {"identifiers": "meter_1541217672", "name": "water_meter_1541217672", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": 1541217672, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.3", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"1541217672_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr/1541217672/attributes", "unique_id": "1541217672_reading", "protocol": "r900", "format": "####.###", "unit_of_measurement": "m\u00b3", "icon": "mdi:water", "device_class": "water"}, "1541217672_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "date", "value_template": "{{ value_json.lastseen }}", "unique_id": "1541217672_lastseen"}}, "state_topic": "rtlamr/1541217672/state", "availability_topic": "rtlamr/status", "qos": 1}
[2025-06-07 14:21:10,603] INFO:Publishing to homeassistant/device/1545088012/config: {"device": {"identifiers": "meter_1545088012", "name": "water_meter_1545088012", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": 1545088012, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.3", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"1545088012_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr/1545088012/attributes", "unique_id": "1545088012_reading", "protocol": "r900", "format": "####.###", "unit_of_measurement": "m\u00b3", "icon": "mdi:water", "device_class": "water"}, "1545088012_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "date", "value_template": "{{ value_json.lastseen }}", "unique_id": "1545088012_lastseen"}}, "state_topic": "rtlamr/1545088012/state", "availability_topic": "rtlamr/status", "qos": 1}
[2025-06-07 14:21:11,604] INFO:Publishing to rtlamr/status: online
[2025-06-07 14:21:11,665] DEBUG:Reseting USB device: 001:002
[2025-06-07 14:21:11,914] INFO:Starting RTL_TCP using: /usr/bin/rtl_tcp -s 2048000 -d 0
[2025-06-07 14:21:12,546] INFO:RTL_TCP started!
[2025-06-07 14:21:12,748] INFO:Starting RTLAMR using: /usr/bin/rtlamr -format=json -server=127.0.0.1:1234 -unique=true -filterid=1543497178,1541217672,1545088012 -msgtype=r900,r900,r900
[2025-06-07 14:24:42,721] CRITICAL:Signal 15 received.

```

**LOG for ver. 2.3.4:**
```
[2025-06-07 14:43:55] Using "/data/options.json" config file
[2025-06-07 14:43:55] Fetching default MQTT configuration from http://supervisor/services/mqtt
[2025-06-07 14:43:55] RTLAMR2MQTT Starting...
[2025-06-07 14:43:55] RTL SDR Device 0bda:2838 found on USB port 001:002 - Index: 0
[2025-06-07 14:43:55] No USB device specified in the config file, using the first found.
[2025-06-07 14:43:55] Configured MQTT sender:
[2025-06-07 14:43:55]  > hostname => core-mosquitto
[2025-06-07 14:43:55]  > port => 1883
[2025-06-07 14:43:55]  > username => addons
[2025-06-07 14:43:55]  > client_id => rtlamr2mqtt
[2025-06-07 14:43:55]  > base_topic => rtlamr
[2025-06-07 14:43:55]  > availability_topic => rtlamr/status
[2025-06-07 14:43:55]  > tls => None
[2025-06-07 14:43:55] Resetting USB device: /dev/bus/usb/001/002
[2025-06-07 14:43:55] Reset sucessful.
[2025-06-07 14:43:55] Sending message to MQTT:
[2025-06-07 14:43:55]  > topic => rtlamr/status
[2025-06-07 14:43:55]  > payload => online
[2025-06-07 14:43:55]  > retain => True
[2025-06-07 14:43:55] Trying to start RTL_TCP: /usr/bin/rtl_tcp  -s 2048000
[2025-06-07 14:43:55] RTL_TCP started with PID 10
[2025-06-07 14:44:00] RTL_TCP is ready to receive connections!
[2025-06-07 14:44:00] Trying to start RTLAMR: /usr/bin/rtlamr -msgtype=r900,r900,r900 -format=json -filterid=1543497178,1541217672,1545088012 -unique=true
[2025-06-07 14:44:00] RTLAMR started with PID 15
[2025-06-07 14:44:00] 14:44:00.904380 decode.go:45: CenterFreq: 912380000
[2025-06-07 14:44:00] 14:44:00.904524 decode.go:46: SampleRate: 2359296
[2025-06-07 14:44:00] 14:44:00.904533 decode.go:47: DataRate: 32768
[2025-06-07 14:44:00] 14:44:00.904540 decode.go:48: ChipLength: 72
[2025-06-07 14:44:00] 14:44:00.904545 decode.go:49: PreambleSymbols: 32
[2025-06-07 14:44:00] 14:44:00.904552 decode.go:50: PreambleLength: 4608
[2025-06-07 14:44:00] 14:44:00.904559 decode.go:51: PacketSymbols: 116
[2025-06-07 14:44:00] 14:44:00.904565 decode.go:52: PacketLength: 16704
[2025-06-07 14:44:00] 14:44:00.904573 decode.go:59: Protocols: r900
[2025-06-07 14:44:00] 14:44:00.904580 decode.go:60: Preambles: 00000000000000001110010101100100
[2025-06-07 14:44:00] 14:44:00.904587 main.go:124: GainCount: 29
[2025-06-07 14:44:33] {"Time":"2025-06-07T14:44:33.91496547-04:00","Offset":0,"Length":0,"Type":"R900","Message":{"ID":1543497178,"Unkn1":163,"NoUse":32,"BackFlow":0,"Consumption":759994,"Unkn3":0,"Leak":0,"LeakNow":0}}
[2025-06-07 14:44:33] Meter "1543497178" - Consumption 0759.994. Sending value to MQTT.
[2025-06-07 14:44:33] Sending MQTT autodiscovery payload to Home Assistant...
[2025-06-07 14:44:33] Sending message to MQTT:

```

**Config:**
```
general:
  sleep_for: 270
  verbosity: debug
  listen_only: false
  tickle_rtl_tcp: false
  device_id: single
mqtt:
  ha_autodiscovery: true
  ha_autodiscovery_topic: homeassistant
  base_topic: rtlamr
  tls_enabled: false
custom_parameters:
  rtltcp: "-s 2048000"
  rtlamr: "-unique=true"
meters:
  - id: 1543497178
    protocol: r900
    name: water_meter_home
    format: "####.###"
    unit_of_measurement: m³
    icon: mdi:water
    device_class: water
  - id: 1541217672
    protocol: r900
    name: water_meter_1541217672
    format: "####.###"
    unit_of_measurement: m³
    icon: mdi:water
    device_class: water
  - id: 1545088012
    protocol: r900
    name: water_meter_1545088012
    format: "####.###"
    unit_of_measurement: m³
    icon: mdi:water
    device_class: water

```

allangood, 2025-06-07:
Can someone help me to troubleshoot this?
On my end, I cannot reproduce the problem, mostly because I don't have a way to test with real devices anymore.

I would like to see the messages inside MQTT, like this:

[image]

Here, I can see all messages:
`homeassistant/device/<meter_id>/config`
and
`rtlamr/<meter_id>/state`
`rtlamr/<meter_id>/attribute`

allangood, 2025-06-07:
Duplicate #288

closed 2025-06-07 by allangood

amordicus, 2025-06-07:
@allangood I have MQTT Explorer running at all time when testing this.
Only once I got something back from the device "[2025-06-07 14:57:54,943] DEBUG:{"Time":"2025-06-07T14:57:54.943311046-04:00","Offset":0,"Length":40960,"Type":"SCM","Message":{"ID":34128163,"Type":11,"TamperPhy":0,"TamperEnc":0,"Consumption":210457,"ChecksumVal":62880}}"

And the MQTT result was:
{
  "device": {
    "identifiers": "meter_34128163",
    "name": "Water_meter",
    "manufacturer": "RTLAMR2MQTT",
    "model": "Smart Meter",
    "sw_version": "1.0",
    "serial_number": 34128163,
    "hw_version": "1.0"
  },
  "origin": {
    "name": "2mqtt",
    "sw_version": "2025.6.3",
    "support_url": "https://github.com/allangood/rtlamr2mqtt"
  },
  "components": {
    "34128163_reading": {
      "platform": "sensor",
      "name": "Reading",
      "value_template": "{{ value_json.reading|float }}",
      "json_attributes_topic": "rtlamr/34128163/attributes",
      "unique_id": "34128163_reading",
      "protocol": "scm",
      "format": "####.###",
      "unit_of_measurement": "m³",
      "icon": "mdi:gauge",
      "device_class": "water",
      "state_class": "total_increasing"
    },
    "34128163_lastseen": {
      "platform": "sensor",
      "name": "Last Seen",
      "device_class": "date",
      "value_template": "{{ value_json.lastseen }}",
      "unique_id": "34128163_lastseen"
    }
  },
  "state_topic": "rtlamr/34128163/state",
  "availability_topic": "rtlamr/status",
  "qos": 1
}

Various result for today (I don't see anything for the time 2025-06-07T14:57:54 ?):

{"device": {"identifiers": "meter_34128163", "name": "Water_meter", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": 34128163, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.3", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"34128163_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr/34128163/attributes", "unique_id": "34128163_reading", "protocol": "scm", "format": "####.###", "unit_of_measurement": "m\u00b3", "icon": "mdi:gauge", "device_class": "water", "state_class": "total_increasing"}, "34128163_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "date", "value_template": "{{ value_json.lastseen }}", "unique_id": "34128163_lastseen"}}, "state_topic": "rtlamr/34128163/state", "availability_topic": "rtlamr/status", "qos": 1}
06/07/2025 3:23:39 PM(-6.68 minutes) 

{"device": {"identifiers": "meter_34128163", "name": "Water_meter", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": 34128163, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.3", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"34128163_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr/34128163/attributes", "unique_id": "34128163_reading", "protocol": "scm", "format": "####.###", "unit_of_measurement": "m\u00b3", "icon": "mdi:gauge", "device_class": "water", "state_class": "total_increasing"}, "34128163_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "date", "value_template": "{{ value_json.lastseen }}", "unique_id": "34128163_lastseen"}}, "state_topic": "rtlamr/34128163/state", "availability_topic": "rtlamr/status", "qos": 1}
06/07/2025 3:18:55 PM(-4.73 minutes) 

{"device": {"identifiers": "meter_34128163", "name": "Water_meter", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": 34128163, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.3", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"34128163_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr/34128163/attributes", "unique_id": "34128163_reading", "protocol": "scm+", "format": "####.###", "unit_of_measurement": "m\u00b3", "icon": "mdi:gauge", "device_class": "water", "state_class": "total_increasing"}, "34128163_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "date", "value_template": "{{ value_json.lastseen }}", "unique_id": "34128163_lastseen"}}, "state_topic": "rtlamr/34128163/state", "availability_topic": "rtlamr/status", "qos": 1}
06/07/2025 2:56:31 PM(-22.41 minutes) 

{"device": {"identifiers": "meter_34128163", "name": "Water_meter", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": 34128163, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.3", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"34128163_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr/34128163/attributes", "unique_id": "34128163_reading", "protocol": "scm", "format": "####.###", "unit_of_measurement": "m\u00b3", "icon": "mdi:gauge", "device_class": "water", "state_class": "total_increasing"}, "34128163_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "date", "value_template": "{{ value_json.lastseen }}", "unique_id": "34128163_lastseen"}}, "state_topic": "rtlamr/34128163/state", "availability_topic": "rtlamr/status", "qos": 1}
06/07/2025 1:56:03 PM(-60.47 minutes) 

Hope this helps.
In HAOS I cannot find anything on how to debug rtl_tcp, if you know of something let me know .