allangood/rtlamr2mqtt #294: update to latest version gives me "ValueError: invalid literal for int() with base 10"

ljmerza, 2025-06-07:
Updated my docker container recently and now im getting an error:

```
2025-06-07T23:10:52.650366396Z [2025-06-07 19:10:52,649] INFO:Publishing to rtlamr/status: online
2025-06-07T23:10:52.667719728Z [2025-06-07 19:10:52,667] DEBUG:Reseting USB device: 0bda:2838
2025-06-07T23:10:52.668254800Z Traceback (most recent call last):
2025-06-07T23:10:52.668673439Z   File "/opt/venv/app/rtlamr2mqtt.py", line 432, in <module>
2025-06-07T23:10:52.668682122Z     main()
2025-06-07T23:10:52.668684313Z     ~~~~^^
2025-06-07T23:10:52.668709174Z   File "/opt/venv/app/rtlamr2mqtt.py", line 333, in main
2025-06-07T23:10:52.668710980Z     rtltcp = start_rtltcp(config)
2025-06-07T23:10:52.668712734Z   File "/opt/venv/app/rtlamr2mqtt.py", line 130, in start_rtltcp
2025-06-07T23:10:52.668714557Z     usbutil.reset_usb_device(usb_id)
2025-06-07T23:10:52.668716228Z     ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
2025-06-07T23:10:52.668718156Z   File "/opt/venv/app/helpers/usb_utils.py", line 48, in reset_usb_device
2025-06-07T23:10:52.668719999Z     busnum, devnum = [int(x) for x in usbdev.split(':')]
2025-06-07T23:10:52.668721701Z                       ~~~^^^
2025-06-07T23:10:52.668723352Z ValueError: invalid literal for int() with base 10: '0bda'
```

```
  rtlamr:
    container_name: rtlamr
    image: allangood/rtlamr2mqtt
    env_file: ./env/global.env
    restart: unless-stopped
    devices:
      - /dev/bus/usb
    volumes:
      - ./volumes/rtlamr/rtlamr2mqtt.yaml:/etc/rtlamr2mqtt.yaml
      - ./volumes/rtlamr/data:/var/lib/rtlamr2mqtt
```

bobwithanx, 2025-06-08:
I'm getting the same error on 2025.6.3

allangood, 2025-06-09:
This is clearly explained in the documentation:
Here: https://github.com/allangood/rtlamr2mqtt?tab=readme-ov-file#multiple-rtl-devices
And here: https://github.com/allangood/rtlamr2mqtt/blob/main/rtlamr2mqtt-addon/DOCS.md#configuration

Please, read the documentation before opening issues.
Also, follow the template to open issues.