allangood/rtlamr2mqtt #311: [Bug]: Home Assistant Auto Discovery is not working

adamderuwe, 2025-06-11:
### What happened?

Data is being published via MQTT to the home assistant autodiscovery topic, but no device is being created in home assistant.

Here is MQTT data:
[image]

Here is a screenshot showing device does not exist in HA:
[image]

Here is the payload of homeassistant/device/69177042/config

homeassistant/device/69177042/config

```{
    "device": {
        "identifiers": "meter_69177042",
        "name": "electric_meter",
        "manufacturer": "RTLAMR2MQTT",
        "model": "Smart Meter",
        "sw_version": "1.0",
        "serial_number": 69177042,
        "hw_version": "1.0"
    },
    "origin": {
        "name": "2mqtt",
        "sw_version": "2025.6.5",
        "support_url": "https://github.com/allangood/rtlamr2mqtt"
    },
    "components": {
        "69177042_reading": {
            "platform": "sensor",
            "name": "Reading",
            "value_template": "{{ value_json.reading|float }}",
            "json_attributes_topic": "rtlamr/69177042/attributes",
            "unique_id": "69177042_reading",
            "protocol": "scm+",
            "format": "#####.##",
            "unit_of_measurement": "kWh",
            "icon": "mdi:gauge",
            "device_class": "energy"
        },
        "69177042_lastseen": {
            "platform": "sensor",
            "name": "Last Seen",
            "device_class": "timestamp",
            "value_template": "{{ value_json.lastseen }}",
            "unique_id": "69177042_lastseen"
        }
    },
    "state_topic": "rtlamr/69177042/state",
    "availability_topic": "rtlamr/status",
    "qos": 1
}
```

### Version used

2025.6.5

### How are you using rtlmr2mqtt?

Home Assitant Add-on

### Configuration file

```shell
general:
  sleep_for: 60
  verbosity: debug
  listen_only: false
  tickle_rtl_tcp: false
custom_parameters:
  rtltcp: "-s 2048000"
  rtlamr: "-unique=true"
mqtt:
  ha_autodiscovery_topic: homeassistant
  ha_status_topic: homeassistant/status
  base_topic: rtlamr
  ha_autodiscovery: true
  tls_enabled: false
  host: 192.168.1.220
  user: username
  password: password
meters:
  - id: 69177042
    protocol: scm+
    name: electric_meter
    format: "#####.##"
    unit_of_measurement: kWh
    icon: mdi:gauge
    device_class: energy
```

### Relevant log output

```shell
[2025-06-11 11:06:58,133] INFO:RTLAMR started!
[2025-06-11 11:10:57,486] INFO:Publishing to rtlamr/69177042/state: {"reading": "75866.43", "lastseen": "2025-06-11T11:10:57-07:00"}
[2025-06-11 11:10:57,486] INFO:Publishing to rtlamr/69177042/attributes: {"FrameSync": 5795, "ProtocolID": 30, "EndpointType": 7, "Tamper": 770, "PacketCRC": 63580, "protocol": "scm+"}
[2025-06-11 11:10:57,486] INFO:All readings received.
[2025-06-11 11:10:57,486] INFO:Sleeping for 60 seconds...
[2025-06-11 11:10:57,486] INFO:Shutting down...
[2025-06-11 11:10:57,486] INFO:Terminating RTLAMR...
[2025-06-11 11:10:57,487] INFO:RTLAMR Terminitaed.
[2025-06-11 11:10:57,487] INFO:Terminating RTL_TCP...
[2025-06-11 11:10:57,489] INFO:RTL_TCP Terminitaed.
[2025-06-11 11:10:57,489] INFO:All done. Bye!
```

reypm, 2025-06-11:
@adamderuwe, don't you need the device_id under `general`? I don't think this add-on is smart enough to know what device_id it should use, see https://github.com/allangood/rtlamr2mqtt/discussions/310

- breaking change: The device_id parameter will fail if it is not formatted in the 000:000 usb device id way

I ran into the same issue, fixed the device_id with the right value, which I believe is `bus:device` ie:

```
Bus 003 Device 010: ID 0bda:2832 Realtek Semiconductor Corp. RTL2832U DVB-T
```
`device_id: '003:010'` and the error is gone, although I am not getting readings

adamderuwe, 2025-06-11:
> [@adamderuwe](https://github.com/adamderuwe) , could you try with the `sleep_for = 0`? I think it is something related to that.

No luck.

[image]

```
general:
  sleep_for: 0
  verbosity: debug
  listen_only: false
  tickle_rtl_tcp: false
custom_parameters:
  rtltcp: "-s 2048000"
  rtlamr: "-unique=true"
mqtt:
  ha_autodiscovery_topic: homeassistant
  ha_status_topic: homeassistant/status
  base_topic: rtlamr
  ha_autodiscovery: true
  tls_enabled: false
  host: 192.168.1.220
  user: user
  password: password
meters:
  - id: 69177042
    protocol: scm+
    name: electric_meter
    format: "#####.##"
    unit_of_measurement: kWh
    icon: mdi:gauge
    device_class: energy

```

```
[2025-06-11 13:41:17,378] INFO:Starting rtlamr2mqtt 2025.6.5
[2025-06-11 13:41:17,378] INFO:Config loaded successfully
[2025-06-11 13:41:17,378] INFO:Connecting to MQTT broker at 192.168.1.220:1883
[2025-06-11 13:41:17,379] INFO:Subscribing to homeassistant/status
[2025-06-11 13:41:17,379] INFO:Publishing to homeassistant/device/69177042/config: {"device": {"identifiers": "meter_69177042", "name": "electric_meter", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": 69177042, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.5", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"69177042_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr/69177042/attributes", "unique_id": "69177042_reading", "protocol": "scm+", "format": "#####.##", "unit_of_measurement": "kWh", "icon": "mdi:gauge", "device_class": "energy"}, "69177042_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "timestamp", "value_template": "{{ value_json.lastseen }}", "unique_id": "69177042_lastseen"}}, "state_topic": "rtlamr/69177042/state", "availability_topic": "rtlamr/status", "qos": 1}
[2025-06-11 13:41:18,380] INFO:Publishing to rtlamr/status: online
[2025-06-11 13:41:18,434] DEBUG:Reseting USB device: 007:002
[2025-06-11 13:41:18,680] INFO:Starting RTL_TCP using: /usr/bin/rtl_tcp -d 0 -s 2048000
[2025-06-11 13:41:18,703] DEBUG:Found 2 device(s):
[2025-06-11 13:41:18,720] DEBUG:0:  Realtek, RTL2838UHIDIR, SN: 00000000
[2025-06-11 13:41:18,738] DEBUG:1:  Realtek, RTL2838UHIDIR, SN: 00000001
[2025-06-11 13:41:18,746] DEBUG:Using device 0: Generic RTL2832U OEM
[2025-06-11 13:41:19,003] DEBUG:Found Rafael Micro R820T tuner
[2025-06-11 13:41:19,185] DEBUG:[R82XX] PLL not locked!
[2025-06-11 13:41:19,259] DEBUG:Tuned to 100000000 Hz.
[2025-06-11 13:41:19,286] DEBUG:listening...
[2025-06-11 13:41:19,286] INFO:RTL_TCP started!
[2025-06-11 13:41:19,487] INFO:Starting RTLAMR using: /usr/bin/rtlamr -filterid=69177042 -msgtype=scm+ -unique=true -format=json -server=127.0.0.1:1234
[2025-06-11 13:41:19,571] DEBUG:13:41:19.570579 decode.go:45: CenterFreq: 912600155
[2025-06-11 13:41:19,571] DEBUG:13:41:19.570759 decode.go:46: SampleRate: 2359296
[2025-06-11 13:41:19,571] DEBUG:13:41:19.570768 decode.go:47: DataRate: 32768
[2025-06-11 13:41:19,571] DEBUG:13:41:19.570801 decode.go:48: ChipLength: 72
[2025-06-11 13:41:19,571] DEBUG:13:41:19.570806 decode.go:49: PreambleSymbols: 16
[2025-06-11 13:41:19,571] DEBUG:13:41:19.570811 decode.go:50: PreambleLength: 2304
[2025-06-11 13:41:19,571] DEBUG:13:41:19.570844 decode.go:51: PacketSymbols: 128
[2025-06-11 13:41:19,571] DEBUG:13:41:19.570850 decode.go:52: PacketLength: 18432
[2025-06-11 13:41:19,571] DEBUG:13:41:19.570857 decode.go:59: Protocols: scm+
[2025-06-11 13:41:19,571] DEBUG:13:41:19.570875 decode.go:60: Preambles: 0001011010100011
[2025-06-11 13:41:19,572] DEBUG:13:41:19.570882 main.go:125: GainCount: 29
[2025-06-11 13:41:19,572] INFO:RTLAMR started!
[2025-06-11 13:41:30,033] INFO:Publishing to rtlamr/69177042/state: {"reading": "75871.05", "lastseen": "2025-06-11T13:41:30-07:00"}
[2025-06-11 13:41:30,034] INFO:Publishing to rtlamr/69177042/attributes: {"FrameSync": 5795, "ProtocolID": 30, "EndpointType": 7, "Tamper": 770, "PacketCRC": 27053, "protocol": "scm+"}
[2025-06-11 13:42:29,256] INFO:Publishing to rtlamr/69177042/state: {"reading": "75871.09", "lastseen": "2025-06-11T13:42:29-07:00"}
[2025-06-11 13:42:29,256] INFO:Publishing to rtlamr/69177042/attributes: {"FrameSync": 5795, "ProtocolID": 30, "EndpointType": 7, "Tamper": 770, "PacketCRC": 46445, "protocol": "scm+"}
[... 7 lines cut ...]
[2025-06-11 13:48:30,089] INFO:Publishing to rtlamr/69177042/state: {"reading": "75871.17", "lastseen": "2025-06-11T13:48:30-07:00"}
[2025-06-11 13:48:30,089] INFO:Publishing to rtlamr/69177042/attributes: {"FrameSync": 5795, "ProtocolID": 30, "EndpointType": 7, "Tamper": 770, "PacketCRC": 7372, "protocol": "scm+"}
[2025-06-11 13:48:30,089] INFO:Publishing to rtlamr/69177042/attributes: {"FrameSync": 5795, "ProtocolID": 30, "EndpointType": 7, "Tamper": 770, "PacketCRC": 7372, "protocol": "scm+"}
[2025-06-11 13:48:30,089] INFO:Publishing to rtlamr/69177042/attributes: {"FrameSync": 5795, "ProtocolID": 30, "EndpointType": 7, "Tamper": 770, "PacketCRC": 7372, "protocol": "scm+"}
[2025-06-11 13:50:32,589] INFO:Publishing to rtlamr/69177042/state: {"reading": "75871.21", "lastseen": "2025-06-11T13:50:32-07:00"}
[2025-06-11 13:50:32,590] INFO:Publishing to rtlamr/69177042/attributes: {"FrameSync": 5795, "ProtocolID": 30, "EndpointType": 7, "Tamper": 770, "PacketCRC": 10958, "protocol": "scm+"}
[... cut ...]
[2025-06-11 14:00:33,201] INFO:Publishing to rtlamr/69177042/attributes: {"FrameSync": 5795, "ProtocolID": 30, "EndpointType": 7, "Tamper": 770, "PacketCRC": 50678, "protocol": "scm+"}
[2025-06-11 14:02:34,591] INFO:Publishing to rtlamr/69177042/attributes: {"FrameSync": 5795, "ProtocolID": 30, "EndpointType": 7, "Tamper": 770, "PacketCRC": 6454, "protocol": "scm+"}
[2025-06-11 14:02:34,591] INFO:Publishing to rtlamr/69177042/attributes: {"FrameSync": 5795, "ProtocolID": 30, "EndpointType": 7, "Tamper": 770, "PacketCRC": 6454, "protocol": "scm+"}
```

reypm, 2025-06-12:
@adamderuwe here ...
```
[2025-06-12 09:16:13,265] CRITICAL:Signal 15 received.
[2025-06-12 09:16:13,266] INFO:Shutting down...
[2025-06-12 09:16:13,266] INFO:Terminating RTLAMR...
[2025-06-12 09:16:13,268] INFO:RTLAMR Terminitaed.
[2025-06-12 09:16:13,268] INFO:Terminating RTL_TCP...
[2025-06-12 09:16:13,270] INFO:RTL_TCP Terminitaed.
[2025-06-12 09:16:13,270] INFO:Publishing to rtlamr/status: offline
[2025-06-12 09:16:13,273] INFO:Disconnecting from MQTT broker
[2025-06-12 09:16:13,274] INFO:All done. Bye!
[2025-06-12 09:16:14,170] INFO:Starting rtlamr2mqtt 2025.6.5
[2025-06-12 09:16:14,170] INFO:Config loaded successfully
[2025-06-12 09:16:14,170] INFO:Connecting to MQTT broker at core-mosquitto:1883
[2025-06-12 09:16:14,172] INFO:Subscribing to hass/status
[2025-06-12 09:16:14,173] INFO:Publishing to homeassistant/device/79263328/config: {"device": {"identifiers": "meter_79263328", "name": "electricity_meter", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": 79263328, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.5", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"79263328_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr/79263328/attributes", "unique_id": "79263328_reading", "protocol": "scm+", "unit_of_measurement": "kWh", "icon": "mdi:meter-electric-outline", "device_class": "energy", "state_class": "total_increasing"}, "79263328_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "timestamp", "value_template": "{{ value_json.lastseen }}", "unique_id": "79263328_lastseen"}}, "state_topic": "rtlamr/79263328/state", "availability_topic": "rtlamr/status", "qos": 1}
[2025-06-12 09:16:14,173] INFO:Publishing to homeassistant/device/39276928/config: {"device": {"identifiers": "meter_39276928", "name": "water_meter", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": 39276928, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.5", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"39276928_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr/39276928/attributes", "unique_id": "39276928_reading", "protocol": "r900", "unit_of_measurement": "gal", "icon": "mdi:meter-water", "device_class": "water", "state_class": "total_increasing"}, "39276928_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "timestamp", "value_template": "{{ value_json.lastseen }}", "unique_id": "39276928_lastseen"}}, "state_topic": "rtlamr/39276928/state", "availability_topic": "rtlamr/status", "qos": 1}
[2025-06-12 09:16:15,173] INFO:Publishing to rtlamr/status: online
[2025-06-12 09:16:15,235] DEBUG:Reseting USB device: 002:004
[2025-06-12 09:16:15,729] INFO:Starting RTL_TCP using: /usr/bin/rtl_tcp -d 0 -s 2048000
[2025-06-12 09:16:15,758] DEBUG:Found 1 device(s):
[2025-06-12 09:16:15,774] DEBUG:0:  Generic, RTL2832U, SN: 77771111153705700
[2025-06-12 09:16:15,779] DEBUG:Using device 0: Generic RTL2832U
[2025-06-12 09:16:16,055] DEBUG:Found Rafael Micro R820T tuner
[2025-06-12 09:16:16,258] DEBUG:[R82XX] PLL not locked!
[2025-06-12 09:16:16,338] DEBUG:Tuned to 100000000 Hz.
[2025-06-12 09:16:16,368] DEBUG:listening...
[2025-06-12 09:16:16,368] INFO:RTL_TCP started!
[2025-06-12 09:16:16,569] INFO:Starting RTLAMR using: /usr/bin/rtlamr -unique=true -server=127.0.0.1:1234 -format=json -filterid=79263328,39276928 -msgtype=scm+,r900
[2025-06-12 09:16:16,661] DEBUG:09:16:16.661309 decode.go:45: CenterFreq: 912600155
[2025-06-12 09:16:16,662] DEBUG:09:16:16.661519 decode.go:46: SampleRate: 2359296
[2025-06-12 09:16:16,662] DEBUG:09:16:16.661530 decode.go:47: DataRate: 32768
[2025-06-12 09:16:16,662] DEBUG:09:16:16.661541 decode.go:48: ChipLength: 72
[2025-06-12 09:16:16,662] DEBUG:09:16:16.661553 decode.go:49: PreambleSymbols: 32
[2025-06-12 09:16:16,662] DEBUG:09:16:16.661561 decode.go:50: PreambleLength: 4608
[2025-06-12 09:16:16,662] DEBUG:09:16:16.661569 decode.go:51: PacketSymbols: 128
[2025-06-12 09:16:16,662] DEBUG:09:16:16.661579 decode.go:52: PacketLength: 18432
[2025-06-12 09:16:16,662] DEBUG:09:16:16.661592 decode.go:59: Protocols: r900,scm+
[2025-06-12 09:16:16,662] DEBUG:09:16:16.661599 decode.go:60: Preambles: 00000000000000001110010101100100,0001011010100011
[2025-06-12 09:16:16,662] DEBUG:09:16:16.661611 main.go:125: GainCount: 29
[2025-06-12 09:16:16,662] INFO:RTLAMR started!
 Live
```

adamderuwe, 2025-06-12:
Besides the initial publishing of the MQTT discovery messages, I don't see that its publishing readings to MQTT.  Are you opening MQTT explorer and then restarting the addon?  Otherwise, you won't see those messages under the "homeassistant/device/" topic.

When you go to Settings -> Integrations -> MQTT -> Devices and filter on "rtl", do you have nothing listed?

[image]

reypm, 2025-06-17:
@allangood @adamderuwe any advice on this? I am still unable to get readings from any of my meters. Is there like a debug option where I can read from all the meters in a radio or something like that?

adamderuwe, 2025-06-18:
This re-write was botched, imho.  Was released too soon.  I'd go back to the "old" version.

Add a new repository in your Home Assistant addon "store" with the URL of:  https://github.com/allangood/rtlamr2mqtt#old

That will let you install version 2.3.4.

reypm, 2025-06-18:
Hey everyone, let's not argue and instead focus on finding a solution. I'm fine with either keeping the current (latest) version and helping as much as I can or reverting to the old version until the new one is more polished.

@allangood , is there a way to run both the old and new versions simultaneously? I prefer the old one because it works well, but I also want to use the new one to help debug potential issues and assist others in finding solutions.

reypm, 2025-06-18:
@allangood  / @adamderuwe, here is the result of my research. Using the following *same* meters config for all the tested add-on:

```
- id: xxxxxxxx (8 digits)
  protocol: scm+
  name: electricity_meter
  unit_of_measurement: kWh
  icon: mdi:meter-electric-outline
  device_class: energy
  state_class: total_increasing
- id: xxxxxxxxx (9 digits)
  protocol: r900
  name: water_meter
  unit_of_measurement: gal
  icon: mdi:meter-water
  device_class: water
  state_class: total_increasing
```

-  Using [AMR2MQTT](https://github.com/mdegat01/addon-amr2mqtt). I get readings from both meters:

[image]

- Using rtlamr2mqtt#old. I get readings from just the electricity meter.
- Using rtlamr2mqtt. I get readings from just the electricity meter.

Let me know if you need any further help

closed 2025-06-20 by allangood