allangood/rtlamr2mqtt #314: [Bug]: 2025.6.5 parsing error from MQTT feed for meters

jcollas, 2025-06-12:
### What happened?

Appears that the parsing code to update the sensors isn't expecting the json payload being sent to MQTT.

Logger: homeassistant.components.mqtt.models
Source: components/mqtt/models.py:366
integration: MQTT (documentation, issues)
First occurred: 10:44:39 PM (9 occurrences)
Last logged: 10:46:50 PM

Exception raised while updating state of sensor.electric_meter, topic: 'rtlamr/XXX/attributes' with payload: b'{"Type": 4, "TamperPhy": 0, "TamperEnc": 3, "ChecksumVal": 46919, "protocol": "scm"}'
Exception raised while updating state of sensor.water_meter, topic: 'rtlamr/XXX/state' with payload: b'{"reading": "01572.87", "lastseen": "2025-06-11T22:44:48-04:00"}'
Exception raised while updating state of sensor.water_meter, topic: 'rtlamr/XXX/attributes' with payload: b'{"Type": 13, "TamperPhy": 2, "TamperEnc": 2, "ChecksumVal": 62578, "protocol": "scm"}'
Exception raised while updating state of sensor.gas_meter, topic: 'rtlamr/XXX/state' with payload: b'{"reading": "04526.68", "lastseen": "2025-06-11T22:46:50-04:00"}'
Exception raised while updating state of sensor.gas_meter, topic: 'rtlamr/XXX/attributes' with payload: b'{"Type": 12, "TamperPhy": 2, "TamperEnc": 0, "ChecksumVal": 8648, "protocol": "scm"}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 666, in state
    numerical_value = float(value)  # type:ignore[arg-type]
ValueError: could not convert string to float: '{"reading": "01572.87", "lastseen": "2025-06-11T22:39:47-04:00"}'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 366, in process_write_state_requests
    entity.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1019, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1141, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1081, in __async_calculate_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1025, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 668, in state
    raise ValueError(
    ...<5 lines>...
    ) from err
ValueError: Sensor sensor.water_meter has device class 'water', state class 'total_increasing' unit 'ft³' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: '{"reading": "01572.87", "lastseen": "2025-06-11T22:39:47-04:00"}' (<class 'str'>)

### How are you using rtlmr2mqtt?

None

### Configuration file

```shell

```

### Relevant log output

```shell

```

jeffdeal, 2025-06-12:
Aha! This is what I'm getting now too, on 2025.6.5 with HA version 2025.6.0.

2025-06-12 10:47:13.584 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised while updating state of sensor.16_water_meter, topic: 'rtlamr/1574941678/attributes' with payload: b'{"Unkn1": 163, "NoUse": 34, "BackFlow": 0, "Unkn3": 0, "Leak": 0, "LeakNow": 0, "protocol": "r900"}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 666, in state
    numerical_value = float(value)  # type:ignore[arg-type]
ValueError: could not convert string to float: '{"reading": "055848.1", "lastseen": "2025-06-12T10:47:13-04:00"}'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 366, in process_write_state_requests
    entity.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1019, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1141, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1081, in __async_calculate_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1025, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 668, in state
    raise ValueError(
    ...<5 lines>...
    ) from err
ValueError: Sensor sensor.16_water_meter has device class 'water', state class 'total_increasing' unit 'gal' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: '{"reading": "055848.1", "lastseen": "2025-06-12T10:47:13-04:00"}' (<class 'str'>)

RogueDruid, 2025-06-12:
> unit_of_measurement

I believe the issue is related to the  _unit_of_measurement: CCF_ 

`  - id: XXX
    protocol: scm
    name: water_meter
    format: "#####.##"
    unit_of_measurement: CCF
    icon: mdi:water
    device_class: water`

per the [HA const.py](https://github.com/home-assistant/core/blob/d7ac4bd65379e11461c7ce0893d3533d8d8b8cbf/homeassistant/const.py#L384) CCF isnt a measure I believe you would want ft³

richardkuw, 2025-06-14:
I am far from an MQTT expert, but what I'm seeing is that the state format has changed. I guess I can tweak my config to compensate but this seems like a breaking change from the previous version.

The energy integration in HA appears to expect the sensor state to have a numeric value. Previously it did, now it does not, it has a json attribute called reading that contains the value. 

Again, I am not an MQTT expert, I have been using HA for years but generally only dig into MQTT minutia when things break.

Log from version 2.3.4:

> [2025-06-09 17:58:53] Sending message to MQTT:
> [2025-06-09 17:58:53]  > topic => rtlamr/35133418/state
> [2025-06-09 17:58:53]  > payload => 0827187
> [2025-06-09 17:58:53]  > retain => True

Log from version 2025.6.5:

> [2025-06-12 08:11:57,496] INFO:Publishing to rtlamr/35133418/state: {"reading": "0827528", "lastseen": "2025-06-12T08:11:57-05:00"}

DaftHonk, 2025-06-20:
You should have two sensors: `sensor.water_meter_reading`and `sensor.water_meter_last_seen`. If you still have a `sensor.water_meter`it is probably left over from version 2.3.4.

jcollas, 2025-06-20:
Yup, when I switched to the _reading meters, everything is working as expected.  I didn't look at the energy dashboard configuration and so didn't notice that the entities had been changed.  My bad, thanks for the hint.  We can probably close this now.

richardkuw, 2025-06-20:
Right. Initially all that was listed in the HA Energy Dashboard configuration dropdown was the old sensor.water_meter. It didn’t recognize the _reading entities. I added state_class: total_increasing to the rtlamr2mqtt meter configuration and it’s working now. That probably should have been there anyway but 2.3.4 worked without it.