allangood/rtlamr2mqtt #318: [Bug]: custom parameters for rtlamr

gregbert42, 2025-06-14:
### What happened?

It doesn't appear to be sleeping after 300s if it doesnt get a reading in this new version.

```
general:
  # Sleep for this amount of seconds after one successful reading of every meter
  # This parameter is helpful to keep CPU usage low and the temperature low as well
  # Set this to 0 (default) to disable it
# 3000s = 50 min
  sleep_for: 3000

  # Set the verbosity level. It can be debug or info
  verbosity: debug

  # Enable/disable the tickle_rtl_tcp. This is used to "shake" rtl_tcp to wake it up.
  # For me, this started to cause the rtl_tcp to refuse connections and miss the readings.
  # This may help with a remote rtl_tcp server.
  tickle_rtl_tcp: false

  # (Optional) USB Device ID. Use lsusb to get the device ID
  # Use "single" (default) if you have only one device
  # device_id: 'single'
  # device_id: '0bda:2838'

device_id: 'single'
custom_parameters:
 rtlamr: '-duration=300s'
```

LOGS:
```

2025-06-14 13:15:50,841] INFO:Starting rtlamr2mqtt 2025.6.5
[2025-06-14 13:15:50,844] INFO:Config loaded successfully
[2025-06-14 13:15:50,844] INFO:Connecting to MQTT broker at 192.168.2.24:1883
[2025-06-14 13:15:50,844] INFO:Subscribing to homeassistant/status
[2025-06-14 13:15:50,845] INFO:Publishing to homeassistant/device/###/config: {"device": <stuff>}
[2025-06-14 13:15:50,845] INFO:Publishing to homeassistant/device/1569614534/config: {"device": {"identifiers": "meter_1569614534", "name": "water_meter", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": 1569614534, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.5", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"1569614534_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr/1569614534/attributes", "unique_id": "1569614534_reading", "protocol": "R900", "format": "#####.##", "unit_of_measurement": "ft\u00b3", "device_class": "water", "state_class": "total_increasing"}, "1569614534_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "timestamp", "value_template": "{{ value_json.lastseen }}", "unique_id": "1569614534_lastseen"}}, "state_topic": "rtlamr/1569614534/state", "availability_topic": "rtlamr/status", "qos": 1}
[2025-06-14 13:15:51,845] INFO:Publishing to rtlamr/status: online
[2025-06-14 13:15:51,871] DEBUG:Reseting USB device: 001:002
[2025-06-14 13:15:52,266] INFO:Starting RTL_TCP using: /usr/bin/rtl_tcp -s 2048000 -d 0
[2025-06-14 13:15:52,288] DEBUG:Found 1 device(s):
[2025-06-14 13:15:52,315] DEBUG:0:  Realtek, RTL2838UHIDIR, SN: 00000001
[2025-06-14 13:15:52,317] DEBUG:Using device 0: Generic RTL2832U OEM
[2025-06-14 13:15:52,745] DEBUG:Found Rafael Micro R820T tuner
[2025-06-14 13:15:53,073] DEBUG:[R82XX] PLL not locked!
[2025-06-14 13:15:53,198] DEBUG:Tuned to 100000000 Hz.
[2025-06-14 13:15:53,252] DEBUG:listening...
[2025-06-14 13:15:53,252] INFO:RTL_TCP started!
[2025-06-14 13:15:53,452] INFO:Starting RTLAMR using: /usr/bin/rtlamr -duration=300s -filterid=75582637,1569614534 -msgtype=scm+,R900 -server=127.0.0.1:1234 -unique=true -format=json
[2025-06-14 13:15:53,601] DEBUG:13:15:53.595198 decode.go:45: CenterFreq: 912380000
[2025-06-14 13:15:53,601] DEBUG:13:15:53.601051 decode.go:46: SampleRate: 2359296
[2025-06-14 13:15:53,601] DEBUG:13:15:53.601056 decode.go:47: DataRate: 32768
[2025-06-14 13:15:53,601] DEBUG:13:15:53.601061 decode.go:48: ChipLength: 72
[2025-06-14 13:15:53,601] DEBUG:13:15:53.601065 decode.go:49: PreambleSymbols: 32
[2025-06-14 13:15:53,601] DEBUG:13:15:53.601069 decode.go:50: PreambleLength: 4608
[2025-06-14 13:15:53,601] DEBUG:13:15:53.601072 decode.go:51: PacketSymbols: 128
[2025-06-14 13:15:53,601] DEBUG:13:15:53.601078 decode.go:52: PacketLength: 18432
[2025-06-14 13:15:53,601] DEBUG:13:15:53.601083 decode.go:59: Protocols: scm+,r900
[2025-06-14 13:15:53,601] DEBUG:13:15:53.601087 decode.go:60: Preambles: 0001011010100011,00000000000000001110010101100100
[2025-06-14 13:15:53,601] DEBUG:13:15:53.601092 main.go:125: GainCount: 29
[2025-06-14 13:15:53,601] INFO:RTLAMR started!
[2025-06-14 13:18:38,468] INFO:Publishing to rtlamr/75582637/state: {"reading": "00331384", "lastseen": "2025-06-14T13:18:38-04:00"}
[2025-06-14 13:18:38,468] INFO:Publishing to rtlamr/75582637/attributes: {"FrameSync": 5795, "ProtocolID": 30, "EndpointType": 156, "Tamper": 2312, "PacketCRC": 35180, "protocol": "scm+"}

```

### How are you using rtlmr2mqtt?

None

### Configuration file

```shell

```

### Relevant log output

```shell

```

davidcoulson, 2025-06-14:
That's not the expected behavior. It will wait indefinitely for a reading then sleep for sleep_for seconds.

gregbert42, 2025-06-14:
> That's not the expected behavior. It will wait indefinitely for a reading then sleep for sleep_for seconds.

not true - at least of  the previous version. This used to make it go to sleep after 300s, regardless of whether or not it had received input from all meters. (in combination with the sleep_for setting)

custom_parameters:
 rtlamr: '-duration=300s'

allangood, 2025-06-20:
https://github.com/allangood/rtlamr2mqtt/pull/320 Address this issue

Ready to be tested on `dev` branch

gregbert42, 2025-06-22:
> [#320](https://github.com/allangood/rtlamr2mqtt/pull/320) Address this issue
> 
> Ready to be tested on `dev` branch

Not what I would expect. note the critical error of rtlamr dying, and then immediate restarting.
I decreased the duration to 30s to test it.

```
[2025-06-22 18:35:21,457] INFO:Starting RTLAMR using: /usr/bin/rtlamr -msgtype=scm+,scm+,R900 -duration=30s -unique=true -filterid=xyz -format=json -server=127.0.0.1:1234
[2025-06-22 18:35:25,474] DEBUG:18:35:25.474366 decode.go:45: CenterFreq: 912380000
[2025-06-22 18:35:25,474] DEBUG:18:35:25.474461 decode.go:46: SampleRate: 2359296
[2025-06-22 18:35:25,474] DEBUG:18:35:25.474468 decode.go:47: DataRate: 32768
[2025-06-22 18:35:25,474] DEBUG:18:35:25.474472 decode.go:48: ChipLength: 72
[2025-06-22 18:35:25,474] DEBUG:18:35:25.474478 decode.go:49: PreambleSymbols: 32
[2025-06-22 18:35:25,474] DEBUG:18:35:25.474483 decode.go:50: PreambleLength: 4608
[2025-06-22 18:35:25,474] DEBUG:18:35:25.474487 decode.go:51: PacketSymbols: 128
[2025-06-22 18:35:25,474] DEBUG:18:35:25.474493 decode.go:52: PacketLength: 18432
[2025-06-22 18:35:25,474] DEBUG:18:35:25.474499 decode.go:59: Protocols: scm+,r900
[2025-06-22 18:35:25,474] DEBUG:18:35:25.474503 decode.go:60: Preambles: 0001011010100011,00000000000000001110010101100100
[2025-06-22 18:35:25,475] DEBUG:18:35:25.474513 main.go:125: GainCount: 29
[2025-06-22 18:35:25,475] INFO:RTLAMR started!
[2025-06-22 18:35:45,477] INFO:Publishing to rtlamr/status: online
[2025-06-22 18:35:45,477] INFO:Publishing to rtlamr/1569614534/state: {"reading": "49662.47", "lastseen": "2025-06-22T18:35:45-04:00"}
[2025-06-22 18:35:45,478] INFO:Publishing to rtlamr/1569614534/attributes: {"Unkn1": 163, "NoUse": 32, "BackFlow": 0, "Unkn3": 0, "Leak": 4, "LeakNow": 0, "protocol": "R900"}
[2025-06-22 18:35:56,479] CRITICAL:RTLAMR has died, trying to restart...
[2025-06-22 18:35:56,679] INFO:Starting RTLAMR using: /usr/bin/rtlamr -msgtype=scm+,scm+,R900 -duration=30s -unique=true -filterid=xyz -format=json -server=127.0.0.1:1234
[2025-06-22 18:36:00,696] DEBUG:18:36:00.695867 decode.go:45: CenterFreq: 912380000
[2025-06-22 18:36:00,696] DEBUG:18:36:00.695953 decode.go:46: SampleRate: 2359296
[2025-06-22 18:36:00,696] DEBUG:18:36:00.695960 decode.go:47: DataRate: 32768
[2025-06-22 18:36:00,696] DEBUG:18:36:00.695964 decode.go:48: ChipLength: 72
[2025-06-22 18:36:00,696] DEBUG:18:36:00.695969 decode.go:49: PreambleSymbols: 32
[2025-06-22 18:36:00,696] DEBUG:18:36:00.695973 decode.go:50: PreambleLength: 4608
[2025-06-22 18:36:00,696] DEBUG:18:36:00.695977 decode.go:51: PacketSymbols: 128
[2025-06-22 18:36:00,696] DEBUG:18:36:00.695981 decode.go:52: PacketLength: 18432
[2025-06-22 18:36:00,696] DEBUG:18:36:00.695986 decode.go:59: Protocols: scm+,r900
[2025-06-22 18:36:00,696] DEBUG:18:36:00.695991 decode.go:60: Preambles: 0001011010100011,00000000000000001110010101100100
[2025-06-22 18:36:00,696] DEBUG:18:36:00.695998 main.go:125: GainCount: 29
[2025-06-22 18:36:00,696] INFO:RTLAMR started!

```

gregbert42, 2025-06-22:
> This is exactly what is expected with this work around.
> The process dies and it is restated.
>
> The sleep_for is meant to sleep *after* a successful reading.

Apologies for not reposting my entire configuration here. But there is
indeed a sleep_for: 3000 in my config file that does not seem to be doing
what I would expect, in combination with that custom parameter to set a
duration of 30 . Let me know if you would like for me to do a complete
repost of my config if that would help you. Thank you.

allangood, 2025-06-25:
It should work as expected now. Could you test and post the result here, please?

gregbert42, 2025-06-26:
yes, it appears the bug has been fixed.. thank you

I am seeing some other issues with other things that used to work as well (home assistant statistics graph) ,but need to try to understand more about whats going on before i post them.

thank you very much

```
Attaching to rtlamr2mqtt
rtlamr2mqtt  | [2025-06-26 17:42:57,850] INFO:Starting rtlamr2mqtt 2025.6.6
rtlamr2mqtt  | [2025-06-26 17:42:57,853] INFO:Config loaded successfully
rtlamr2mqtt  | [2025-06-26 17:42:57,853] INFO:Connecting to MQTT broker at 192.168.2.24:1883
rtlamr2mqtt  | [2025-06-26 17:42:57,853] INFO:Subscribing to homeassistant/status
rtlamr2mqtt  | [2025-06-26 17:42:57,854] INFO:Publishing to homeassistant/device/stuff
rtlamr2mqtt  | [2025-06-26 17:42:57,854] INFO:Publishing to homeassistant/device/stuff
rtlamr2mqtt  | [2025-06-26 17:42:58,854] INFO:Publishing to rtlamr/status: online
rtlamr2mqtt  | [2025-06-26 17:42:58,878] DEBUG:Reseting USB device: 001:002
rtlamr2mqtt  | [2025-06-26 17:42:59,275] INFO:Starting RTL_TCP using: /usr/bin/rtl_tcp -s 2048000 -d 0
rtlamr2mqtt  | [2025-06-26 17:42:59,299] DEBUG:Found 1 device(s):
rtlamr2mqtt  | [2025-06-26 17:42:59,325] DEBUG:0:  Realtek, RTL2838UHIDIR, SN: 00000001
rtlamr2mqtt  | [2025-06-26 17:42:59,328] DEBUG:Using device 0: Generic RTL2832U OEM
rtlamr2mqtt  | [2025-06-26 17:42:59,756] DEBUG:Found Rafael Micro R820T tuner
rtlamr2mqtt  | [2025-06-26 17:43:00,084] DEBUG:[R82XX] PLL not locked!
rtlamr2mqtt  | [2025-06-26 17:43:00,209] DEBUG:Tuned to 100000000 Hz.
rtlamr2mqtt  | [2025-06-26 17:43:00,263] DEBUG:listening...
rtlamr2mqtt  | [2025-06-26 17:43:00,263] INFO:RTL_TCP has started!
rtlamr2mqtt  | [2025-06-26 17:43:00,464] INFO:Starting RTLAMR using: /usr/bin/rtlamr -unique=true -format=json -duration=3s -server=127.0.0.1:1234 -filterid=75582637,1569614534 -msgtype=scm+,R900
rtlamr2mqtt  | [2025-06-26 17:43:00,612] DEBUG:17:43:00.606691 decode.go:45: CenterFreq: 912380000
rtlamr2mqtt  | [2025-06-26 17:43:00,612] DEBUG:17:43:00.612626 decode.go:46: SampleRate: 2359296
rtlamr2mqtt  | [2025-06-26 17:43:00,612] DEBUG:17:43:00.612631 decode.go:47: DataRate: 32768
rtlamr2mqtt  | [2025-06-26 17:43:00,612] DEBUG:17:43:00.612636 decode.go:48: ChipLength: 72
rtlamr2mqtt  | [2025-06-26 17:43:00,612] DEBUG:17:43:00.612642 decode.go:49: PreambleSymbols: 32
rtlamr2mqtt  | [2025-06-26 17:43:00,612] DEBUG:17:43:00.612645 decode.go:50: PreambleLength: 4608
rtlamr2mqtt  | [2025-06-26 17:43:00,612] DEBUG:17:43:00.612650 decode.go:51: PacketSymbols: 128
rtlamr2mqtt  | [2025-06-26 17:43:00,612] DEBUG:17:43:00.612653 decode.go:52: PacketLength: 18432
rtlamr2mqtt  | [2025-06-26 17:43:00,612] DEBUG:17:43:00.612659 decode.go:59: Protocols: scm+,r900
rtlamr2mqtt  | [2025-06-26 17:43:00,612] DEBUG:17:43:00.612664 decode.go:60: Preambles: 0001011010100011,00000000000000001110010101100100
rtlamr2mqtt  | [2025-06-26 17:43:00,612] DEBUG:17:43:00.612668 main.go:125: GainCount: 29
rtlamr2mqtt  | [2025-06-26 17:43:00,612] INFO:RTLAMR has started!
rtlamr2mqtt  | [2025-06-26 17:43:04,613] CRITICAL:RTLAMR has died, trying to restart...
rtlamr2mqtt  | [2025-06-26 17:43:04,613] INFO:Sleep for is set to 3000 seconds...

```

closed 2025-06-26 by gregbert42