allangood/rtlamr2mqtt #321: [Bug]: Messages not being publish to MQTT

reypm, 2025-06-21:
### What happened?

I keep trying to get this working, but so far, no luck, messages are not being published to MQTT, and I am not sure why at this point. This is how my configuration looks:

```yaml
general:
  sleep_for: 0
  verbosity: debug
  listen_only: false
  tickle_rtl_tcp: false
mqtt:
  ha_autodiscovery_topic: homeassistant
  ha_status_topic: hass/status
  base_topic: rtlamr
  host: core-mosquitto
  port: 1883
  user: mqtt_user
  password: <redacted>
  ha_autodiscovery: true
  tls_enabled: false
custom_parameters:
  rtltcp: "-s 2048000"
  rtlamr: "-unique=true"
meters:
  - id: <meter_id>
    protocol: scm+
    name: electricity_meter
    unit_of_measurement: kWh
    icon: mdi:meter-electric-outline
    device_class: energy
    state_class: total_increasing

```
MQTT integration shows both sensors with `Unknown` status:

[image]

Here is the debug log from the add-on:

```
[2025-06-21 09:17:47,426] INFO:Starting rtlamr2mqtt 2025.6.5
[2025-06-21 09:17:47,427] INFO:Config loaded successfully
[2025-06-21 09:17:47,427] INFO:Connecting to MQTT broker at core-mosquitto:1883
[2025-06-21 09:17:47,428] INFO:Subscribing to hass/status
[2025-06-21 09:17:47,429] INFO:Publishing to homeassistant/device/69963948/config: {"device": {"identifiers": "meter_69963948", "name": "electricity_meter", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": 69963948, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.5", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"69963948_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr/69963948/attributes", "unique_id": "69963948_reading", "protocol": "scm+", "unit_of_measurement": "kWh", "icon": "mdi:meter-electric-outline", "device_class": "energy", "state_class": "total_increasing"}, "69963948_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "timestamp", "value_template": "{{ value_json.lastseen }}", "unique_id": "69963948_lastseen"}}, "state_topic": "rtlamr/69963948/state", "availability_topic": "rtlamr/status", "qos": 1}
[2025-06-21 09:17:48,429] INFO:Publishing to rtlamr/status: online
[2025-06-21 09:17:48,483] DEBUG:Reseting USB device: 002:004
[2025-06-21 09:17:48,977] INFO:Starting RTL_TCP using: /usr/bin/rtl_tcp -d 0 -s 2048000
[2025-06-21 09:17:49,001] DEBUG:Found 1 device(s):
[2025-06-21 09:17:49,016] DEBUG:0:  Generic, RTL2832U, SN: 77771111153705700
[2025-06-21 09:17:49,020] DEBUG:Using device 0: Generic RTL2832U
[2025-06-21 09:17:49,283] DEBUG:Found Rafael Micro R820T tuner
[2025-06-21 09:17:49,476] DEBUG:[R82XX] PLL not locked!
[2025-06-21 09:17:49,550] DEBUG:Tuned to 100000000 Hz.
[2025-06-21 09:17:49,578] DEBUG:listening...
[2025-06-21 09:17:49,579] INFO:RTL_TCP started!
[2025-06-21 09:17:49,780] INFO:Starting RTLAMR using: /usr/bin/rtlamr -msgtype=scm+ -format=json -server=127.0.0.1:1234 -unique=true -filterid=69963948
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870619 decode.go:45: CenterFreq: 912600155
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870803 decode.go:46: SampleRate: 2359296
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870814 decode.go:47: DataRate: 32768
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870862 decode.go:48: ChipLength: 72
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870878 decode.go:49: PreambleSymbols: 16
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870885 decode.go:50: PreambleLength: 2304
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870891 decode.go:51: PacketSymbols: 128
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870903 decode.go:52: PacketLength: 18432
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870929 decode.go:59: Protocols: scm+
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870936 decode.go:60: Preambles: 0001011010100011
[2025-06-21 09:17:49,872] DEBUG:09:17:49.870949 main.go:125: GainCount: 29
[2025-06-21 09:17:49,872] INFO:RTLAMR started!
```

what I am missing here?

### How are you using rtlmr2mqtt?

Home Assitant Add-on

### Configuration file

```shell
general:
  sleep_for: 0
  verbosity: debug
  listen_only: false
  tickle_rtl_tcp: false
mqtt:
  ha_autodiscovery_topic: homeassistant
  ha_status_topic: hass/status
  base_topic: rtlamr
  host: core-mosquitto
  port: 1883
  user: mqtt_user
  password: <redacted>
  ha_autodiscovery: true
  tls_enabled: false
custom_parameters:
  rtltcp: "-s 2048000"
  rtlamr: "-unique=true"
meters:
  - id: <meter_id>
    protocol: scm+
    name: electricity_meter
    unit_of_measurement: kWh
    icon: mdi:meter-electric-outline
    device_class: energy
    state_class: total_increasing
```

### Relevant log output

```shell
[2025-06-21 09:17:47,426] INFO:Starting rtlamr2mqtt 2025.6.5
[2025-06-21 09:17:47,427] INFO:Config loaded successfully
[2025-06-21 09:17:47,427] INFO:Connecting to MQTT broker at core-mosquitto:1883
[2025-06-21 09:17:47,428] INFO:Subscribing to hass/status
[2025-06-21 09:17:47,429] INFO:Publishing to homeassistant/device/69963948/config: {"device": {"identifiers": "meter_69963948", "name": "electricity_meter", "manufacturer": "RTLAMR2MQTT", "model": "Smart Meter", "sw_version": "1.0", "serial_number": 69963948, "hw_version": "1.0"}, "origin": {"name": "2mqtt", "sw_version": "2025.6.5", "support_url": "https://github.com/allangood/rtlamr2mqtt"}, "components": {"69963948_reading": {"platform": "sensor", "name": "Reading", "value_template": "{{ value_json.reading|float }}", "json_attributes_topic": "rtlamr/69963948/attributes", "unique_id": "69963948_reading", "protocol": "scm+", "unit_of_measurement": "kWh", "icon": "mdi:meter-electric-outline", "device_class": "energy", "state_class": "total_increasing"}, "69963948_lastseen": {"platform": "sensor", "name": "Last Seen", "device_class": "timestamp", "value_template": "{{ value_json.lastseen }}", "unique_id": "69963948_lastseen"}}, "state_topic": "rtlamr/69963948/state", "availability_topic": "rtlamr/status", "qos": 1}
[2025-06-21 09:17:48,429] INFO:Publishing to rtlamr/status: online
[2025-06-21 09:17:48,483] DEBUG:Reseting USB device: 002:004
[2025-06-21 09:17:48,977] INFO:Starting RTL_TCP using: /usr/bin/rtl_tcp -d 0 -s 2048000
[2025-06-21 09:17:49,001] DEBUG:Found 1 device(s):
[2025-06-21 09:17:49,016] DEBUG:0:  Generic, RTL2832U, SN: 77771111153705700
[2025-06-21 09:17:49,020] DEBUG:Using device 0: Generic RTL2832U
[2025-06-21 09:17:49,283] DEBUG:Found Rafael Micro R820T tuner
[2025-06-21 09:17:49,476] DEBUG:[R82XX] PLL not locked!
[2025-06-21 09:17:49,550] DEBUG:Tuned to 100000000 Hz.
[2025-06-21 09:17:49,578] DEBUG:listening...
[2025-06-21 09:17:49,579] INFO:RTL_TCP started!
[2025-06-21 09:17:49,780] INFO:Starting RTLAMR using: /usr/bin/rtlamr -msgtype=scm+ -format=json -server=127.0.0.1:1234 -unique=true -filterid=69963948
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870619 decode.go:45: CenterFreq: 912600155
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870803 decode.go:46: SampleRate: 2359296
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870814 decode.go:47: DataRate: 32768
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870862 decode.go:48: ChipLength: 72
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870878 decode.go:49: PreambleSymbols: 16
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870885 decode.go:50: PreambleLength: 2304
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870891 decode.go:51: PacketSymbols: 128
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870903 decode.go:52: PacketLength: 18432
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870929 decode.go:59: Protocols: scm+
[2025-06-21 09:17:49,871] DEBUG:09:17:49.870936 decode.go:60: Preambles: 0001011010100011
[2025-06-21 09:17:49,872] DEBUG:09:17:49.870949 main.go:125: GainCount: 29
[2025-06-21 09:17:49,872] INFO:RTLAMR started!
```

reypm, 2025-06-22:
And that is exactly my problem, why? All config seems ok to me using a different add-on (amr2mqtt) I get readings from the metter

allangood, 2025-06-22:
This is probably an error in your configuration.
If you are absolutely sure about the ID, you can try at the `protocol` to `all` to check if it can catch the id

reypm, 2025-06-23:
That does not work :) 

> Failed to save add-on configuration, value must be one of ['idm', 'netidm', 'r900', 'r900bcd', 'scm', 'scm+']

allangood, 2025-06-25:
Right... I keep forgetting about the add-ons restrictions. Sorry.

If you are comfortable with command line, you can try to run the container manually:
```
docker run --rm -ti --entrypoint /bin/bash --device /dev/bus:/dev/bus allangood/rtlamr2mqtt:dev
rtl_tcp &
rtlamr -msgtype=all 
```

This will give you an output like this:
```
~$ docker run --rm -ti --entrypoint /bin/bash --device /dev/bus:/dev/bus allangood/rtlamr2mqtt:dev
root@c40ce6db45e3:/# rtl_tcp &
[1] 7
root@c40ce6db45e3:/# Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Tuned to 100000000 Hz.
listening...
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).

root@c40ce6db45e3:/# rtlamr -msgtype=all 
client accepted! localhost 37006
Allocating 15 zero-copy buffers
set freq 912380000
20:57:32.790851 decode.go:45: CenterFreq: 912380000
20:57:32.791043 decode.go:46: SampleRate: 2359296
20:57:32.791058 decode.go:47: DataRate: 32768
20:57:32.791068 decode.go:48: ChipLength: 72
20:57:32.791077 decode.go:49: PreambleSymbols: 32
20:57:32.791087 decode.go:50: PreambleLength: 4608
20:57:32.791097 decode.go:51: PacketSymbols: 736
20:57:32.791107 decode.go:52: PacketLength: 105984
20:57:32.791123 decode.go:59: Protocols: scm,scm+,idm,r900
20:57:32.791150 decode.go:60: Preambles: 01010101010101010001011010100011,00000000000000001110010101100100,111110010101001100000,0001011010100011
20:57:32.791186 main.go:125: GainCount: 29
set sample rate 2359296
set gain mode 0

```

After this, all meters from your neighborhood will show up there and you can confirm your ID and Protocol

closed 2025-07-07 by allangood

reypm, 2025-07-23:
nvm, is working as expected, great update on the latest release

chucknorris101, 2025-08-29:
@allangood I just got this search function to run on my CLI to try to identify why my gas meter wasnt showing up...but have a followon question

i see my water meter reading fine in the addon, and i see it popping up in the cli as it scrolls past, but I dont see any other meter information (in dense residential...) - i just get lots of 11+, 1....502 lines that are blank apart from where my water meter pops up

 Is this a configuration issue, or like my receiver isnt picking up any other signals? the receiver is about as far from the water as the gas, so not sure why one would come through perfectly and the other not at all.