allangood/rtlamr2mqtt #325: [Bug]: New version stopped using existing HA entities

jwaters23, 2025-06-26:
### What happened?

I haven't been following the code rewrites here, I just noticed that after an update to my docker container that is implementing this project, my existing Home Assistant entities for my gas and electric meters stopped updating. I now see new MQTT devices in HA with "sensor.unknown_meter_reading" and "sensor.unknown_meter_reading_2" entities and those are being updated correctly. Why did these new entities get created? Why did the existing entities stop getting updated? How do I fix things so that my existing entities I've been using for years are updated and not these new ones that got created somehow?

### Version used

2026.6.5

### How are you using rtlmr2mqtt?

Stand alone Docker

### Configuration file

```shell
# -- Configuration file starts here --
# (Optional section)
general:
  # Sleep for this number of seconds after one successful reading of every meter
  # This parameter is helpful to keep CPU usage low and the temperature low as well
  # Set this to 0 (default) to disable it
  sleep_for: 60
  # Set the verbosity level. It can be debug or info
  verbosity: debug
  # Enable/disable the tickle_rtl_tcp. This is used to "shake" rtl_tcp to wake it up.
  # For me, this started to cause the rtl_tcp to refuse connections and miss the readings.
  # This may help with a remote rtl_tcp server.
  tickle_rtl_tcp: false
  # (Optional) USB Device ID. Use lsusb to get the device ID
  # Use "single" (default) if you have only one device
  # device_id: 'single'
  # device_id: '002/003'

# MQTT configuration.
mqtt:
  # Whether to use Home Assistant auto-discovery feature or not
  ha_autodiscovery: false
  # Home Assistant auto-discovery topic
  ha_autodiscovery_topic: homeassistant
  # Base topic to send status and updates
  base_topic: rtlamr
  # By default, leaving host, port, user, and password unset will tell
  # rtlamr2mqtt to use the default home assistant mqtt settings for those
  # options. If needed, you can override these default settings:
  # MQTT host name or IP address.
  host: <ip_redacted>
  # MQTT port.
  port: 1883
  # TLS Enabled? (False by default)
  tls_enabled: false
  # TLS CA certificate (mandatory if tls_enabled = true)
  tls_ca: "/etc/ssl/certs/ca-certificates.crt"
  # TLS server certificate (optional)
  tls_cert: "/etc/ssl/my_server_cert.crt"
  # TLS self-signed certificate/insecure certificate (optional, default true)
  tls_insecure: true
  # MQTT user name if you have, remove if you don't use authentication
  #user: mqtt
  # MQTT user password if you use one, remove if you don't use authentication
  #password: my-very-strong-password

# (Optional)
# This entire section is optional.
# If you don't need any custom parameter, don't use it.
# ***DO NOT ADD -msgtype, -filterid nor -protocol parameters here***
# -d parameter is not necessary anymore if you use device_id
#custom_parameters:
  # Documentation for rtl_tcp: https://osmocom.org/projects/rtl-sdr/wiki/Rtl-sdr
  #rtltcp: "-s 2048000"
  # Documentation for rtlamr: https://github.com/bemasher/rtlamr/wiki/Configuration
  # If you want to disable the local rtl_tcp and use an external/remote one, you must add "-server=remote-ip-address:port" to the rtlamr section below.
  #rtlamr: "-version=true"

# (Required section)
# Here is the place to define your meters
meters:
    # The ID of your meter
  - id: <redacted>
    # The protocol
    protocol: scm
    # A nice name to show on your Home Assistant/Node Red
    #name: gas_meter_new
    # (optional) A number format to be used for your meter
    format: "####.##"
    # (optional) A measurement unit to be used by Home Assistant
    # Typical values are ft³ and m³ (use the superscript) for water/gas meters
    # and kWh or Wh for electric meters
    #unit_of_measurement: "CCF"
    # (optional) An icon to be used by Home Assistant
    #icon: mdi:gauge
    # A device_class to define what the sensor is measuring for use in the Energy panel
    # Typical values are "gas" or "energy". Default is blank.
    #device_class: gas
    # "total_increasing" for most meters, "total" for meters that might go
    # backwards (net energy meters). Defaults to "total_increasing" if unset.
    #state_class: total_increasing
  - id: <redacted> 
    protocol: scm+
    format: "#####.##"
# -- End of configuration file --
```

### Relevant log output

```shell

```

allangood, 2025-06-27:
It is a new sensor/entity. Isn't possible to "migrate" the data.
You can try updating your database (sqlite db) and have you data showing in the new entity, but I don't recommend it unless you know what you are doing.

The utility meter integration will not loose any history if you change the sensor associated to it:
https://www.home-assistant.io/integrations/utility_meter/

jwaters23, 2025-06-29:
OK thanks for clarifying. In the previous system I had both the RTLAMR configuration file that I posted here, and also created an MQTT sensor in my main home assistant configuration. Do I no longer need the MQTT sensor defined separately from the RTLAMR configuration?

Here were my sensors in mqtt_sensors.yaml:

- name: electric_meter_net
  state_topic: "rtlamr/<redacted>/state"
  unit_of_measurement: 'kWh'
  expire_after: 600
  device_class: energy
  state_class: total_increasing
  unique_id: rtlamr_electric_meter_net_kwh
  
- name: gas_meter
  state_topic: "rtlamr/<redacted>/state"
  unit_of_measurement: 'CCF'
  device_class: gas
  state_class: total_increasing
  unique_id: rtlamr_gas_CCF

allangood, 2025-06-29:
Your manually created sensor aren't necessary in any of the versions, old or new in reality. 

You only need the utility meter integration besides this addon

jwaters23, 2025-06-29:
OK I updated my RTLAMR configuration just to be more complete with information. I don't see it as an available sensor in my HA Energy configuration dashboard:

# (Required section)
# Here is the place to define your meters
meters:
    # The ID of your meter
  - id: <redacted>
    # The protocol
    protocol: scm
    # A nice name to show on your Home Assistant/Node Red
    name: gas_meter
    # (optional) A number format to be used for your meter
    format: "####.##"
    # (optional) A measurement unit to be used by Home Assistant
    # Typical values are ft³ and m³ (use the superscript) for water/gas meters
    # and kWh or Wh for electric meters
    unit_of_measurement: "CCF"
    # (optional) An icon to be used by Home Assistant
    #icon: mdi:gauge
    # A device_class to define what the sensor is measuring for use in the Energy panel
    # Typical values are "gas" or "energy". Default is blank.
    device_class: gas
    # "total_increasing" for most meters, "total" for meters that might go
    # backwards (net energy meters). Defaults to "total_increasing" if unset.
    state_class: total_increasing
  - id: <redacted>
    name: electric_net_meter 
    protocol: scm+
    format: "#####.##"
    unit_of_measurement: "kWh"
    device_class: energy
    state_class: total
# -- End of configuration file --