allangood/rtlamr2mqtt #327: MQTT message retention has changed in the new version.

flexnsniff, 2025-06-28:
Thanks for your time building a wonderful little tool! 

With the old version of rtlamr2mqtt, the mqtt messages were retained - allowing me to subscribe to the relevant topics with Telegraf, and export them to InfluxDB. While I also use rtlamr2mqtt to use the data inside of Homeassistant, I rely upon the telegraf export of the mqtt data to influxDB for visualization inside of Graphana. 

When looking and the old and new codebase, I noticed that the old version uses a mixture of retain=True and retain=False, but the new version uses retain=False throughout the code. 

It'd be great if the retain value could be specified in the mqtt: section of the rtlamr2mqtt configuration.

allangood, 2025-06-30:
While I understand the user-case, I still don't think it is necessary to make changes in the code as you can achieve the same results using native ways.

Retaining messages isn't a good thing as it will leave the data even when nothing is consuming it or after a crash. It opens to a myriad of problems and "it is not working" type of requests here.

You can achieve the same results using Home Assistant InfluxDB integration:
https://www.home-assistant.io/integrations/influxdb/

Just export the sensor to your InfluxDB and you will not need Telegraf to collect the data from MQTT.

flexnsniff, 2025-06-30:
Absolutely understand why hardcoding “true” could be a problem; thus why a default of “false” but configurable “true” would be my only consideration.

I understand that Homeassistant includes a InfluxDB export - I'm already using it. However, HA doesn’t import the mqtt data as often as it’s published. If I want more frequent updates, getting the data directly from mqtt negates the sparse, random updates that HA subscribes to.  This allows me to have more detailed usage of my house power - once every 5 minutes as the default in rtlamr2mqtt - making it easier to identify usage trends.

allangood, 2025-06-30:
I think adding a flag in the config to allow retention would be acceptable as the person change it must know what this means.

A lot of the bug requests I have been receiving for this new version is related to the retained messages from the previous version. This is something I want to avoid to happen again.