allangood/rtlamr2mqtt #331: Help With Adding R900 Meter As Water Source Which Will Show Up On Energy Dashboard

nicklesb8787, 2025-07-06:
I can see the meter value change within logs but no matter what I try I can not get it to show up in energy dashboard.

rtlamr2mqtt yaml config:

general:
  sleep_for: 60
  verbosity: info
custom_parameters: {}
mqtt:
  ha_autodiscovery_topic: homeassistant
  ha_status_topic: homeassistant/status
  base_topic: rtlamr
meters:
  - id: 1572885774
    protocol: r900
    name: water_meter
    format: "######.###"
    device_class: water

log:
[2025-07-06 18:25:11,821] DEBUG:Reseting USB device: 001:003
[2025-07-06 18:25:12,186] INFO:Starting RTL_TCP using: /usr/bin/rtl_tcp -d 0 -s 2048000
[2025-07-06 18:25:13,169] INFO:RTL_TCP has started!
[2025-07-06 18:25:13,371] INFO:Starting RTLAMR using: /usr/bin/rtlamr -msgtype=r900 -server=127.0.0.1:1234 -unique=true -filterid=1572885774 -format=json
[2025-07-06 18:25:13,511] INFO:RTLAMR has started!
[2025-07-06 18:25:52,533] INFO:Publishing to rtlamr/status: online
[2025-07-06 18:25:52,534] INFO:Publishing to rtlamr/1572885774/state: {"reading": "008051.500", "lastseen": "2025-07-06T18:25:52-04:00"}
[2025-07-06 18:25:52,534] INFO:Publishing to rtlamr/1572885774/attributes: {"Unkn1": 163, "NoUse": 0, "BackFlow": 0, "Unkn3": 0, "Leak": 0, "LeakNow": 0, "protocol": "r900"}
[2025-07-06 18:25:52,535] INFO:All readings received.
[2025-07-06 18:25:52,535] INFO:Sleeping for 60 seconds...
[2025-07-06 18:25:52,536] INFO:Shutting down...
[2025-07-06 18:25:52,536] INFO:Terminating RTLAMR...
[2025-07-06 18:25:52,538] INFO:RTLAMR Terminitaed.
[2025-07-06 18:25:52,538] INFO:Terminating RTL_TCP...
[2025-07-06 18:25:52,544] INFO:RTL_TCP Terminitaed.
[2025-07-06 18:25:52,544] INFO:All done. Bye!
[2025-07-06 18:26:52,545] INFO:Time to wake up!
[2025-07-06 18:26:53,546] CRITICAL:RTL_TCP has died, trying to restart...
[2025-07-06 18:26:53,549] DEBUG:Reseting USB device: 001:003
[2025-07-06 18:26:53,743] INFO:Starting RTL_TCP using: /usr/bin/rtl_tcp -d 0 -s 2048000
[2025-07-06 18:26:54,324] INFO:RTL_TCP has started!
[2025-07-06 18:26:54,324] CRITICAL:RTLAMR has died, trying to restart...
[2025-07-06 18:26:54,325] INFO:Sleep for is set to 60 seconds...

Where it has "reading" 8051.500 is the number of gal I use and I see that number change, but no matter what I do I can not get it to reflect as a sensor or entity. Please help thank you for your time and assistance.

Respectfully,
Brad

allangood, 2025-07-07:
How is your utility meter integration configured?

nicklesb8787, 2025-07-07:
Do you mean what is in my configuation.yaml?

DaftHonk, 2025-07-08:
Add `state_class: total_increasing` to your meter config.

nicklesb8787, 2025-07-09:
I see it in logs

[2025-07-08 20:08:07,755] INFO:RTLAMR has started!
[2025-07-08 20:08:21,766] INFO:Publishing to rtlamr/status: online
[2025-07-08 20:08:21,768] INFO:Publishing to rtlamr/1572885774/state: {"reading": "008061.000", "lastseen": "2025-07-08T20:08:21-04:00"}
[2025-07-08 20:08:21,769] INFO:Publishing to rtlamr/1572885774/attributes: {"Unkn1": 163, "NoUse": 0, "BackFlow": 0, "Unkn3": 0, "Leak": 0, "LeakNow": 0, "protocol": "r900"}
[2025-07-08 20:08:21,770] INFO:All readings received.
[2025-07-08 20:08:21,770] INFO:Sleeping for 60 seconds...
[2025-07-08 20:08:21,771] INFO:Shutting down...
[2025-07-08 20:08:21,771] INFO:Terminating RTLAMR...
[2025-07-08 20:08:21,774] INFO:RTLAMR Terminitaed.
[2025-07-08 20:08:21,775] INFO:Terminating RTL_TCP...
[2025-07-08 20:08:21,779] INFO:RTL_TCP Terminitaed.
[2025-07-08 20:08:21,779] INFO:All done. Bye!
[2025-07-08 20:09:21,780] INFO:Time to wake up!
[2025-07-08 20:09:22,780] CRITICAL:RTL_TCP has died, trying to restart...
[2025-07-08 20:09:22,784] DEBUG:Reseting USB device: 001:003
[2025-07-08 20:09:22,975] INFO:Starting RTL_TCP using: /usr/bin/rtl_tcp -d 0 -s 2048000
[2025-07-08 20:09:23,022] DEBUG:Found 1 device(s):
[2025-07-08 20:09:23,038] DEBUG:0:  Realtek, RTL2838UHIDIR, SN: 00000001
[2025-07-08 20:09:23,046] DEBUG:Using device 0: Generic RTL2832U OEM
[2025-07-08 20:09:23,286] DEBUG:Found Rafael Micro R820T tuner
[2025-07-08 20:09:23,459] DEBUG:[R82XX] PLL not locked!
[2025-07-08 20:09:23,528] DEBUG:Tuned to 100000000 Hz.
[2025-07-08 20:09:23,553] DEBUG:listening...
[2025-07-08 20:09:23,553] INFO:RTL_TCP has started!
[2025-07-08 20:09:23,553] CRITICAL:RTLAMR has died, trying to restart...
[2025-07-08 20:09:23,554] INFO:Sleep for is set to 60 seconds...

nicklesb8787, 2025-07-09:
Nothing shows up when I go to energy dashboard and try to select a meter unless I manually edit the configuation.yaml

allangood, 2025-07-09:
I am closing this as it is a "request for help" not a bug. 

You didn't posted your utility meter integration config and is the most important information. 

Please, read Home Assistant utility meter integration. 

Regards

nicklesb8787, 2025-07-09:
I tried, your read me clearly isnt that good if TONS of people have problems getting it configured. Why closed this when clearly you wont help setup your addon?

nicklesb8787, 2025-07-09:
Makes a graph on MQTT explorer but I have no idea how to connect it to utility meter, maybe you should include that in your documents to prevent tons of people having the same questions. Closing is a DICK move.

nicklesb8787, 2025-07-09:
Why would you close the thread? You kidding me? You didnt provide any help
at all.