allangood/rtlamr2mqtt #338: [Bug]: Unable to set "-unique=false" as a custom parameter for rtlamr command line

pointful, 2025-08-03:
### What happened?

I'd like to get all meter reports via rtlamr, even when the consumption value hasn't changed, so that I know the meter is still sending and the value really hasn't changed. So I am trying to set "-unique=false" (which is the default for rtlamr, I think) but it looks like a "-unique=true" gets tacked onto the end of the command string regardless, overriding this setting.

I have this in my "rtlamr2mqtt.yaml" file:
```
custom_parameters:
  rtlamr: "-unique=false"
```

But in the logs, I see this:
`rtlamr2mqtt  | [2025-08-03 02:06:01,422] INFO:Starting RTLAMR using: /usr/bin/rtlamr -server=127.0.0.1:1234 -filterid=XXXXXXXX,XXXXXXXXX,XXXXXXXX -msgtype=scm,scm+,scm -format=json -unique=false -unique=true
`

### Version used

2025.6.6

### How are you using rtlmr2mqtt?

Stand alone Docker

### Configuration file

```shell
general:
  sleep_for: 0
  verbosity: debug

mqtt:
  host: XXX
  port: 1883
  user: XXX
  password: XXX
  tls_enabled: false
  tls_insecure: true
  ha_autodiscovery_topic: homeassistant
  ha_status_topic: homeassistant/status
  base_topic: "rtlamr"

custom_parameters:
  rtlamr: "-unique=false"

meters:
  - id: XXXXXXXX
    protocol: scm
    name: water_meter
    format: "######.##"
    unit_of_measurement: "ft³"
    icon: mdi:gauge
    device_class: water
    state_class: total_increasing
    expire_after: 600
    force_update: true
  - id: XXXXXXXX
    protocol: scm+
    name: water_meter_isabell
    format: "######.##"
    unit_of_measurement: "ft³"
    icon: mdi:gauge
    device_class: water
    state_class: total_increasing
    expire_after: 5400
    force_update: true
  - id: XXXXXXXX
    protocol: scm
    name: gas_meter
    format: "########"
    unit_of_measurement: "ft³"
    icon: mdi:gauge
    device_class: gas
    state_class: total_increasing
    expire_after: 600
    force_update: true
```

### Relevant log output

```shell
rtlamr2mqtt  | [2025-08-03 02:05:59,318] INFO:Starting rtlamr2mqtt 2025.6.6
rtlamr2mqtt  | [2025-08-03 02:05:59,322] INFO:Config loaded successfully
...
rtlamr2mqtt  | [2025-08-03 02:06:00,630] INFO:Starting RTL_TCP using: /usr/bin/rtl_tcp -s 2048000 -d 0
rtlamr2mqtt  | [2025-08-03 02:06:00,695] DEBUG:Found 1 device(s):
rtlamr2mqtt  | [2025-08-03 02:06:00,711] DEBUG:0:  Realtek, RTL2838UHIDIR, SN: 00000001
rtlamr2mqtt  | [2025-08-03 02:06:00,716] DEBUG:Using device 0: Generic RTL2832U OEM
rtlamr2mqtt  | [2025-08-03 02:06:00,953] DEBUG:Found Rafael Micro R820T tuner
rtlamr2mqtt  | [2025-08-03 02:06:01,127] DEBUG:[R82XX] PLL not locked!
rtlamr2mqtt  | [2025-08-03 02:06:01,195] DEBUG:Tuned to 100000000 Hz.
rtlamr2mqtt  | [2025-08-03 02:06:01,221] DEBUG:listening...
rtlamr2mqtt  | [2025-08-03 02:06:01,221] INFO:RTL_TCP has started!
rtlamr2mqtt  | [2025-08-03 02:06:01,422] INFO:Starting RTLAMR using: /usr/bin/rtlamr -server=127.0.0.1:1234 -filterid=XXXXXXXX,XXXXXXXX,XXXXXXXX -msgtype=scm,scm+,scm -format=json -unique=false -unique=true
rtlamr2mqtt  | [2025-08-03 02:06:01,506] DEBUG:02:06:01.506351 decode.go:45: CenterFreq: 912600155
rtlamr2mqtt  | [2025-08-03 02:06:01,506] DEBUG:02:06:01.506474 decode.go:46: SampleRate: 2359296
rtlamr2mqtt  | [2025-08-03 02:06:01,506] DEBUG:02:06:01.506483 decode.go:47: DataRate: 32768
rtlamr2mqtt  | [2025-08-03 02:06:01,506] DEBUG:02:06:01.506487 decode.go:48: ChipLength: 72
rtlamr2mqtt  | [2025-08-03 02:06:01,507] DEBUG:02:06:01.506490 decode.go:49: PreambleSymbols: 21
rtlamr2mqtt  | [2025-08-03 02:06:01,507] DEBUG:02:06:01.506493 decode.go:50: PreambleLength: 3024
rtlamr2mqtt  | [2025-08-03 02:06:01,507] DEBUG:02:06:01.506499 decode.go:51: PacketSymbols: 128
rtlamr2mqtt  | [2025-08-03 02:06:01,507] DEBUG:02:06:01.506503 decode.go:52: PacketLength: 18432
rtlamr2mqtt  | [2025-08-03 02:06:01,507] DEBUG:02:06:01.506510 decode.go:59: Protocols: scm,scm+
rtlamr2mqtt  | [2025-08-03 02:06:01,507] DEBUG:02:06:01.506516 decode.go:60: Preambles: 111110010101001100000,0001011010100011
rtlamr2mqtt  | [2025-08-03 02:06:01,507] DEBUG:02:06:01.506528 main.go:125: GainCount: 29
rtlamr2mqtt  | [2025-08-03 02:06:01,507] INFO:RTLAMR has started!
```

allangood, 2025-08-03:
Duplicate #337

closed 2025-08-03 by allangood

pointful, 2025-08-03:
Sorry for the duplicate. Thanks again!

allangood, 2025-08-03:
Thank you for taking time to reporting!