allangood/rtlamr2mqtt #341: [Bug]: Defunct processes left behind

chennin, 2025-08-05:
### What happened?

I found that every successful measurement leaves behind 2 defunct processes, inflating `ps` output and taking up pids. Maybe only an issue with `sleep_for` set?

```bash
# ps -ef | grep defunc[t]
rtlamr2+  315969  315929  0 10:11 ?        00:00:00 [rtl_tcp] <defunct>
rtlamr2+  315991  315929  1 10:12 ?        00:00:02 [rtlamr] <defunct>
rtlamr2+  316508  315929  0 10:13 ?        00:00:00 [rtl_tcp] <defunct>
rtlamr2+  316821  315929  5 10:14 ?        00:00:01 [rtlamr] <defunct>
```

I don't know why it's happening nor the details of Python subprocessing, but I do have a docker fix, which is to use dumb-init as the entrypoint:

```diff
diff --git a/rtlamr2mqtt-addon/Dockerfile b/rtlamr2mqtt-addon/Dockerfile
index a541161..b3932b9 100644
--- a/rtlamr2mqtt-addon/Dockerfile
+++ b/rtlamr2mqtt-addon/Dockerfile
@@ -29,6 +29,7 @@ RUN apt-get update \
     && apt-get install -o Dpkg::Options::="--force-confnew" -y \
       libusb-1.0-0 \
       expect \
+      dumb-init \
     && apt-get --purge autoremove -y \
     && apt-get clean \
     && find /var/lib/apt/lists/ -type f -delete \
@@ -37,4 +38,5 @@ RUN apt-get update \

 STOPSIGNAL SIGTERM

-ENTRYPOINT ["python", "/opt/venv/app/rtlamr2mqtt.py"]
+ENTRYPOINT ["dumb-init"]
+CMD ["python", "/opt/venv/app/rtlamr2mqtt.py"]
```

---

Mentioning my docker config in the event it's relevant at all:

```yaml
  rtlamr2mqtt:
    container_name: prod-rtlamr2mqtt
    image: allangood/rtlamr2mqtt
    restart: unless-stopped
    user: 2006:46
    devices:
      - '/dev/rtlsdr:/dev/bus/usb/003/003'
    volumes:
      - '/root/etc-docker/rtlamr2mqtt/rtlamr2mqtt.yaml:/etc/rtlamr2mqtt.yaml'
      - '/etc/localtime:/etc/localtime:ro'
      - '/etc/timezone:/etc/timezone:ro'
    logging:
      <<: *logging
```

### Version used

2025.6.6

### How are you using rtlmr2mqtt?

Stand alone Docker

### Configuration file

```shell
general:
  sleep_for: 60
  verbosity: debug

mqtt:
  host: redacted.net
  port: 5649
  user: rtlsdr
  password: redacted
  tls_enabled: true
  tls_insecure: false
  tls_ca: "/etc/ssl/certs/ca-certificates.crt"
  ha_autodiscovery_topic: homeassistant
  ha_status_topic: homeassistant/status
  base_topic: "rtlamr"

meters:
  - id: 1234567890
    protocol: r900
    name: water_meter
    format: "#######.##"
    unit_of_measurement: "ft³"
    icon: mdi:gauge
    device_class: water
    force_update: true
```

### Relevant log output

```shell
[2025-08-05 10:11:59,179] INFO:Publishing to rtlamr/status: online
[2025-08-05 10:11:59,208] DEBUG:Reseting USB device: 003:003
[2025-08-05 10:11:59,697] INFO:Starting RTL_TCP using: /usr/bin/rtl_tcp -d 0 -s 2048000
[2025-08-05 10:11:59,719] DEBUG:Found 1 device(s):
[2025-08-05 10:11:59,733] DEBUG:0:  Realtek, RTL2838UHIDIR, SN: 00000001
[2025-08-05 10:11:59,737] DEBUG:Using device 0: Generic RTL2832U OEM
[2025-08-05 10:11:59,993] DEBUG:Found Rafael Micro R820T tuner
[2025-08-05 10:12:00,175] DEBUG:[R82XX] PLL not locked!
[2025-08-05 10:12:00,249] DEBUG:Tuned to 100000000 Hz.
[2025-08-05 10:12:00,276] DEBUG:listening...
[2025-08-05 10:12:00,276] INFO:RTL_TCP has started!
[2025-08-05 10:12:00,477] INFO:Starting RTLAMR using: /usr/bin/rtlamr -msgtype=r900 -server=127.0.0.1:1234 -filterid=1234567890 -unique=true -format=json
[2025-08-05 10:12:00,559] DEBUG:10:12:00.559330 decode.go:45: CenterFreq: 912380000
[2025-08-05 10:12:00,559] DEBUG:10:12:00.559607 decode.go:46: SampleRate: 2359296
[2025-08-05 10:12:00,559] DEBUG:10:12:00.559611 decode.go:47: DataRate: 32768
[2025-08-05 10:12:00,559] DEBUG:10:12:00.559614 decode.go:48: ChipLength: 72
[2025-08-05 10:12:00,559] DEBUG:10:12:00.559617 decode.go:49: PreambleSymbols: 32
[2025-08-05 10:12:00,559] DEBUG:10:12:00.559620 decode.go:50: PreambleLength: 4608
[2025-08-05 10:12:00,559] DEBUG:10:12:00.559623 decode.go:51: PacketSymbols: 116
[2025-08-05 10:12:00,559] DEBUG:10:12:00.559626 decode.go:52: PacketLength: 16704
[2025-08-05 10:12:00,559] DEBUG:10:12:00.559631 decode.go:59: Protocols: r900
[2025-08-05 10:12:00,559] DEBUG:10:12:00.559634 decode.go:60: Preambles: 00000000000000001110010101100100
[2025-08-05 10:12:00,559] DEBUG:10:12:00.559640 main.go:125: GainCount: 29
[2025-08-05 10:12:00,559] INFO:RTLAMR has started!
[2025-08-05 10:12:27,565] INFO:Publishing to rtlamr/status: online
[2025-08-05 10:12:27,566] INFO:Publishing to rtlamr/1234567890/state: {"reading": "0001046.66", "lastseen": "2025-08-05T10:12:27-04:00"}
[2025-08-05 10:12:27,566] INFO:Publishing to rtlamr/1234567890/attributes: {"Unkn1": 163, "NoUse": 32, "BackFlow": 0, "Unkn3": 0, "Leak": 0, "LeakNow": 0, "protocol": "r900"}
[2025-08-05 10:12:27,566] INFO:All readings received.
[2025-08-05 10:12:27,566] INFO:Sleeping for 60 seconds...
[2025-08-05 10:12:27,566] INFO:Shutting down...
[2025-08-05 10:12:27,566] INFO:Terminating RTLAMR...
[2025-08-05 10:12:27,567] INFO:RTLAMR Terminitaed.
[2025-08-05 10:12:27,567] INFO:Terminating RTL_TCP...
[2025-08-05 10:12:27,568] INFO:RTL_TCP Terminitaed.
[2025-08-05 10:12:27,568] INFO:All done. Bye!
[2025-08-05 10:13:27,569] INFO:Time to wake up!
[2025-08-05 10:13:28,569] CRITICAL:RTL_TCP has died, trying to restart...
[2025-08-05 10:13:28,570] DEBUG:Reseting USB device: 003:003
[2025-08-05 10:13:29,048] INFO:Starting RTL_TCP using: /usr/bin/rtl_tcp -d 0 -s 2048000
[2025-08-05 10:13:29,060] DEBUG:Found 1 device(s):
[2025-08-05 10:13:29,075] DEBUG:0:  Realtek, RTL2838UHIDIR, SN: 00000001
[2025-08-05 10:13:29,079] DEBUG:Using device 0: Generic RTL2832U OEM
[2025-08-05 10:13:29,335] DEBUG:Found Rafael Micro R820T tuner
[2025-08-05 10:13:29,517] DEBUG:[R82XX] PLL not locked!
[2025-08-05 10:13:29,590] DEBUG:Tuned to 100000000 Hz.
[2025-08-05 10:13:29,618] DEBUG:listening...
[2025-08-05 10:13:29,618] INFO:RTL_TCP has started!
[2025-08-05 10:13:29,618] CRITICAL:RTLAMR has died, trying to restart...
[2025-08-05 10:13:29,618] INFO:Sleep for is set to 60 seconds...
[2025-08-05 10:14:29,819] INFO:Starting RTLAMR using: /usr/bin/rtlamr -msgtype=r900 -server=127.0.0.1:1234 -filterid=1234567890 -unique=true -format=json
[2025-08-05 10:14:29,901] DEBUG:10:14:29.901499 decode.go:45: CenterFreq: 912380000
[2025-08-05 10:14:29,901] DEBUG:10:14:29.901579 decode.go:46: SampleRate: 2359296
[2025-08-05 10:14:29,901] DEBUG:10:14:29.901586 decode.go:47: DataRate: 32768
[2025-08-05 10:14:29,901] DEBUG:10:14:29.901591 decode.go:48: ChipLength: 72
[2025-08-05 10:14:29,901] DEBUG:10:14:29.901596 decode.go:49: PreambleSymbols: 32
[2025-08-05 10:14:29,901] DEBUG:10:14:29.901601 decode.go:50: PreambleLength: 4608
[2025-08-05 10:14:29,901] DEBUG:10:14:29.901605 decode.go:51: PacketSymbols: 116
[2025-08-05 10:14:29,901] DEBUG:10:14:29.901610 decode.go:52: PacketLength: 16704
[2025-08-05 10:14:29,901] DEBUG:10:14:29.901615 decode.go:59: Protocols: r900
[2025-08-05 10:14:29,901] DEBUG:10:14:29.901621 decode.go:60: Preambles: 00000000000000001110010101100100
[2025-08-05 10:14:29,901] DEBUG:10:14:29.901626 main.go:125: GainCount: 29
[2025-08-05 10:14:29,901] INFO:RTLAMR has started!
[2025-08-05 10:14:47,905] INFO:Publishing to rtlamr/status: online
[2025-08-05 10:14:47,905] INFO:Publishing to rtlamr/1234567890/state: {"reading": "0001046.66", "lastseen": "2025-08-05T10:14:47-04:00"}
[2025-08-05 10:14:47,905] INFO:Publishing to rtlamr/1234567890/attributes: {"Unkn1": 163, "NoUse": 32, "BackFlow": 0, "Unkn3": 0, "Leak": 0, "LeakNow": 0, "protocol": "r900"}
[2025-08-05 10:14:47,905] INFO:All readings received.
[2025-08-05 10:14:47,905] INFO:Sleeping for 60 seconds...
[2025-08-05 10:14:47,905] INFO:Shutting down...
[2025-08-05 10:14:47,905] INFO:Terminating RTLAMR...
[2025-08-05 10:14:47,906] INFO:RTLAMR Terminitaed.
[2025-08-05 10:14:47,906] INFO:Terminating RTL_TCP...
[2025-08-05 10:14:47,907] INFO:RTL_TCP Terminitaed.
[2025-08-05 10:14:47,907] INFO:All done. Bye!
```

allangood, 2025-08-08:
This seems a quick way to fix it.
Subprocess is the easiest way to get insane 😅

I think we could go this route until I figure the python native way

chennin, 2025-08-14:
Closed by #345

closed 2025-08-14 by chennin

brent20, 2025-08-17:
Thank you for this! I'm having the same issue.

PR #345 (2025-08-09): Set Docker entrypoints to dumb-init so it can reap defunct children
Fixes #341

Use dumb-init to reap defunct subprocesses instead of leaving a bunch of <defunct> PIDs around