allangood/rtlamr2mqtt #343: [Bug]:

ghxstxch, 2025-08-07:
### What happened?

Try to be more descriptive than 'it doesn't work' or 'it is broken'.

Container restarts itself every handful of seconds with the attached below dump in the logs: 

Already nuked container, deleted and re-pulled image, tried with new directory, etc.

### Version used

2025.06.06

### How are you using rtlmr2mqtt?

Stand alone Docker

### Configuration file

```shell
version: "3"
services:
  rtlamr:
    container_name: rtlamr2mqtt
    image: allangood/rtlamr2mqtt
    restart: unless-stopped
    devices:
      - /dev/bus/usb
    volumes:
      - ${STRG_PATH}/rtlamr2mqtt/rtlamr2mqtt.yaml:/etc/rtlamr2mqtt.yaml:ro
      - ${STRG_PATH}/rtlamr2mqtt/data:/var/lib/rtlamr2mqtt
networks: {}
```

### Relevant log output

```shell
rtlamr2mqtt  | [2025-08-07 02:58:00,032] DEBUG:Reseting USB device: 0bda:2838
rtlamr2mqtt  | Traceback (most recent call last):
rtlamr2mqtt  |   File "/opt/venv/app/rtlamr2mqtt.py", line 508, in <module>
rtlamr2mqtt  |     main()
rtlamr2mqtt  |     ~~~~^^
rtlamr2mqtt  |   File "/opt/venv/app/rtlamr2mqtt.py", line 344, in main
rtlamr2mqtt  |     rtltcp = start_rtltcp(config)
rtlamr2mqtt  |   File "/opt/venv/app/rtlamr2mqtt.py", line 122, in start_rtltcp
rtlamr2mqtt  |     usbutil.reset_usb_device(usb_id)
rtlamr2mqtt  |     ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
rtlamr2mqtt  |   File "/opt/venv/app/helpers/usb_utils.py", line 48, in reset_usb_device
rtlamr2mqtt  |     busnum, devnum = [int(x) for x in usbdev.split(':')]
rtlamr2mqtt  |                       ~~~^^^
rtlamr2mqtt  | ValueError: invalid literal for int() with base 10: '0bda'
```

allangood, 2025-08-07:
https://github.com/allangood/rtlamr2mqtt/discussions/310
https://github.com/allangood/rtlamr2mqtt/issues/316

ghxstxch, 2025-08-09:
I missed this, I am so sorry for opening up a unnecessary bug report and wasting your time!

allangood, 2025-08-09:
No need to apologize.
Than you