allangood/rtlamr2mqtt #356: [Bug]: Meter ID as a hexadecimal string not integer in rtlamr2mqtt add on

DFWJim, 2025-09-29:
### What happened?

My electric meter company has Landis+Gyr Gridstream RF meters.  The meter is placed by a lines-provider which is separate from the billing company.  I ran rtl_433 and listened to and found my meter.  The meter's physical label has a hexadecimal number (example only): f0123ab1 and a integer at the bottom.  When I pull the meter data from my electric company's API, I use the numeric integer (9 digits).  That is allowed to be saved in the configuraton for rtlamr2mqtt.  however when attempting to use the hexadecimal number, it errors with the following error:
Failed to save add-on configuration, expected int. 

Yet setting it to the integer works and is allowed to be saved.  When receiving the radio signal, the id field is the hex number.  I get no readings from rtlamr2mqtt when using the numeric integer id number.

We need to be able to use the hex number in the id field of the yaml validator in HA.
When using the 9 digit integer, it works.  Documentation says put my meter id number in the id: field.

### Version used

2025.6.6

### How are you using rtlmr2mqtt?

Home Assitant Add-on

### Configuration file

```shell
general:
  sleep_for: 60
  verbosity: info
custom_parameters:
  rtltcp: "-s 2M -centerfreq=926M"
  rtlamr: ""
mqtt:
  ha_autodiscovery_topic: homeassistant
  ha_status_topic: homeassistant/status
  base_topic: homeassistant
meters:
  - id: 123456789
    protocol: scm
    name: My Electric Meter
    format: "######.###"
    device_class: energy
    center_frequency: 926000000
```

### Relevant log output

```shell
Failed to save add-on configuration, expected int. Got {'general': {'sleep_for': 60, 'verbosity': 'info'}, 'custom_parameters': {'rtltcp': '-s 2M -centerfreq=926M', 'rtlamr': ''}, 'mqtt': {'ha_autodiscovery_topic': 'homeassistant', 'ha_status_topic': 'homeassistant/status', 'base_topic': 'homeassistant'}, 'meters': [{'id': '123456789', 'protocol': 'scm', 'name': 'My Electric Meter', 'format': '######.###', 'device_class': 'energy', 'center_frequency': 926000000}]}
```

DFWJim, 2025-09-29:
double quotes, single quotes, and no quotes all have the same result.

allangood, 2026-04-22:
Long shot, but is this still relevant?